lgsvl/simulator #72: Intial position of Ego vehicle Apollo - LG simulator

kk2491, 2019-01-16:
Hi,

I am currently working on Apollo and was able to setup LG simulator with Apollo by following the given steps. I am using San Fransico map to run the simulation. 
I have some queries on the simulator, kindly help me in understanding the same.

**1. Initial position of the Ego vehicle**
When I start the simulation does initial position of the Ego vehicle in Dreamview matches with the position of the vehicle in LG simulator. As far as I noticed, it looks like the position of the Ego vehicle is different considering the layout of the road.
If both are different, how would I reset the position in LG simulator so that it matches with Dreamview.

**2. Routing Start and End point**
While I was running the simulator I faced the below issue. 
Lets consider the position of Ego vehicle in LG simulator be (X1, Y1), now I give a routing request in Dreamview with start point as (X2, Y2) and end point as (X3, Y3). 
In LG simulator, does the car actually start the routing from (X2, Y2) ? or from (X1, Y1) ?
I noticed that in LG simulator it doesnt reset the position to start point of the routing request, rather it starts directly from the current position in LG simulator.

**3. Custom map**
Is there a way to create map (looks like map editor is disabled)?
Is there a tool which can be used to convert Apollo HD map directly into LG simulator map? 

Kindly let me know if you need more details from my end to clarify my queries.

Thanks in advance.

KK

hadiTab, 2019-01-16:
2. In Apollo if you pass the routing module multiple points it will find a route between them. The position of the ego vehicle is not taken into consideration and does not reset the vehicle position. Apollo doesn't know that it is driving the ego vehicle in the simulator, it thinks it's driving a real vehicle. So it can't simply reset the vehicle position to the start of the route.

3. At the moment the only way to create a custom map would be to manually build it in the unity editor. There are some instructions for this in https://github.com/lgsvl/simulator/issues/60.  As for directly building off of an HD map, this is something we are looking into; however, I do not have a timeline for it's completion.

Hope this answers your questions.

hadiTab, 2019-01-16:
@kk2491 I'm not sure what exactly you mean by unstable, but if you do have more than one instance of the simulator running (I do this a lot by accident) the car in dreamview will repeatedly jump between the location of the ego vehicle in each of the two simulator instances.

kk2491, 2019-01-21:
@hadiTab Thanks for the reply. 
Unstable means there was some fluctuation in simulation. 
In dreamview simulation Ego vehicle jumps from one location to another. And there are no 2 similar simulator instances are running in the system.

Thank you,
KK

hadiTab, 2019-01-21:
@kk2491 what you are describing happens when either two different localization topics are being published, or when two different GNSS topics are being published. If you experience it again please check the nodes publishing these topics by doing:
```
rostopic info /apollo/localization/pose
rostopic info /apollo/sensors/gnss/best_pose
```
Note that these commands must be executed inside the apollo docker container.
Please share the results here as well.
Aside from having two instances of the simulator running the only other way I can think of this happening is if you have a rosbag playing at the same time as the simulator is running.

kk2491, 2019-01-22:
@hadiTab .. Thanks for the answer.

I do not have any rosbag playing during the simulation. 

Please find the output of ```rostopic info``` for the mentioned topics.
```
kishor@in_dev_docker:/apollo$ rostopic info /apollo/localization/pose
Type: pb_msgs/LocalizationEstimate

Publishers: 
 * /dreamview (http://in_dev_docker:43301/)
 * /localization (http://in_dev_docker:36211/)

Subscribers: 
 * /SystemMonitor (http://in_dev_docker:37935/)
 * /dreamview (http://in_dev_docker:43301/)
 * /perception (http://in_dev_docker:45697/)
 * /std_planning (http://in_dev_docker:41331/)
 * /prediction (http://in_dev_docker:41143/)
 * /control (http://in_dev_docker:33225/)

kishor@in_dev_docker:/apollo$ rostopic info /apollo/sensor/gnss/best_pose
Type: pb_msgs/GnssBestPose

Publishers: None

Subscribers: 
 * /SystemMonitor (http://in_dev_docker:37935/)

kishor@in_dev_docker:/apollo$ 
```

In localization pose, I could see 2 publisher. Is that the issue causing the fluctuation ?

I am running several simulation to understand the issue and fix. Will update you if I find any useful information. 

Thank you,
KK

hadiTab, 2019-01-22:
@kk2491 it might be. The only way I can think of dreamview publishing localization would be if "simControl" was enabled. Please make sure that it is not enabled.

hadiTab, 2019-01-28:
@kk2491 after starting the localization and routing modules the HD map should load and you should then be able to access the route editing page (without needing to enable/disable simControl).

kk2491, 2019-01-29:
@hadiTab Yea. I missed it. Thanks for the information.

kk2491, 2019-02-06:
Yea.. Thank you so much for the help.

closed 2019-02-06 by kk2491

xiejingjacob, 2022-07-05:
Hi, I meet the same problem with initial position incorrect. How can fix this problem?

Thanks