lgsvl/simulator #75: How to use multiple Lidars in the single vehicle.

wsp50317, 2019-01-19:
I tried to directly copy the Lidar and run, but the copied one would be stuck in the original position(It can't move with the vehicle). And It also failed to pass data to Autoware.
Is there any methods to use multi Lidars?

moenasrallah, 2019-01-21:
I think the problem you are having is that the Lidar is not a child object to the Car object.
Observe this relationship and drag the new copied Lidar to be like this in the hierarchy.

[image]

wsp50317, 2019-01-24:
@moenasrallah Sorry that I didn't explain exactly. I have set parent and child, and the Lidars can move with vehicle. But the clone one's red point cloud would stuck in the start point in the unity scene.

@martins-mozeiko  Thank you! It solve my problem. Although the above problem have not be solved, ROS works normally.

martins-mozeiko, 2019-01-24:
@wsp50317 You are right. Visualization in simulator does not work correctly when multiple lidars are added to vehicle. We will fix this.

martins-mozeiko, 2019-01-30:
Visualization is fixed with e669702 commit.

closed 2019-01-30 by martins-mozeiko