lgsvl/simulator #76: rosbridge performance

yangydavid, 2019-01-22:
Hello LG developers, 

We are running LG simulator on one machine at 60fps (we only collect GPS+IMU data) and on the other machine we run only the ros bridge (without Apollo). We recorded a rosbag to check the localization precision ("apollo/localization/pose" topic). We have a couple of questions regrading to the logs we obtained, as shown in the picture below (position x vs time):
[image]
1) On the image you can spot an outlier - could it be a simulator glitch or it is a network bug?
2) As you can see there is 0.5s data drop. What could be the reason?
3) I noticed that the sampling rate is quite low - the bridge spits out the data with ~100Hz but it appears that the data is being updated only with ~10Hz rate. Can you please clarify the sampling rate situation?
4) The x position axis resolution is quite low - roughly 0.5m. What could be the reason for such a low resolution?

martins-mozeiko, 2019-01-22:
Hello.

When I run simulator on Windows machine and Apollo on Linux machine then I see ~12Hz for GPS and ~100Hz for IMU messages. GPS frequency is controlled by `Frequency` property on GPS sensor object. IMU is sent once per every Unity physics update step which currently is 0.01 seconds.

I checked frequencies with `rostopic hz` commands:

    rostopic hz /apollo/sensor/gnss/corrected_imu        
    rostopic hz /apollo/sensor/gnss/best_pose

`/apollo/localization/pose` topic comes out from Apollo. It is using some kind of fusion on other sensors, so I'm not sure what frequency it should be publishing.

Note that we are simulating GPS on a bit higher level than actual NovAtel which is connected as a hardware device to Apollo. It may be that we need adjust frequency of ROS message publish rate for this.

I think 0.5 precision is expected with ~12Hz publish rate. If car moves 20mph, that is ~9m/s. With 12Hz it means that you get 9/12 = 0.75 meter change on every update.

Not sure about the data drop, I'll need to look into this a bit more.

yangydavid, 2019-01-24:
Hi,

thanks for the clarifications. I have the same observations. A couple of follow-up questions:
- how can we modify the GPS sensor object? Can it be done only from Unity Editor or there is a way to configure it after the binary is compiled?
- we attribute the severe package drops (in the magnitude of seconds) to our PC performance. We tried to on a more powerful PC and the situation significantly improved, but we still occasionally encounter package drops/delay in the magnitude of 0.1s (picture below). Please let us know if you have any insights into what could be the reason: network, PC performance or smth else.
Btw, we are using the same setup as you: LG sim on a Windows PC and Apollo on an Unbuntu machine.

[image]

Thanks

Engietoday, 2019-02-07:
> Hello.
>
> When I run simulator on Windows machine and Apollo on Linux machine then I see ~12Hz for GPS and ~100Hz for IMU messages. GPS frequency is controlled by `Frequency` property on GPS sensor [...]
Hello

Can you provide some information on how you got the sim to run on Windows and and it publish things on Ubuntu side? I tried running rosbridge on my Ubuntu while connected to a Windows machine via ethernet but was not able to get the simulator connected.

closed 2020-04-07 by EricBoiseLGSVL