lgsvl/simulator #77: how to make waypoint.

hakuturu583, 2019-01-22:
I read this document(https://www.lgsvlsimulator.com/docs/map-annotation/, https://github.com/lgsvl/simulator/issues/60#issuecomment-451261249) and try map annotation.
But it was failed when I click create tmp waypoint.
No waypoint was shown in the scene view.

[image]

hakuturu583, 2019-01-22:
No error log appears in the console.

luqiang21, 2019-01-22:
> No error log appears in the console.

The waypoint should be shown in the center of the scene. You can try to zoom in since the waypoint may be hard to find and create a temp waypoint again.

luqiang21, 2019-01-22:
> No error log appears in the console.

Please make sure your road belongs to the layer of `Ground and Road`. Waypoint is only created on this layer.

hakuturu583, 2019-02-17:
Thanks!! I succeed!!

closed 2019-02-17 by hakuturu583