lgsvl/simulator #80: [Apollo & Simulator] Cannot choose vehicle and load maps in apollo

huyichen19950417, 2019-01-25:
I have tried with the instructions from your website[https://www.lgsvlsimulator.com/docs/apollo-instructions/](url). I've built up Apollo successfully and try to launch apollo alongside the simulator. Here are several problems I have met with:

- No `HD Mode` in the simulator:[image]

- Cannot select vehicle in Dreamview: [image]
According to the documents, the vehicle `LG SCL 1080` should be selected

- Although the map SanFrancisco is selected, it is not loaded in Dreamview.

Is there any idea to solve these problems?

martins-mozeiko, 2019-01-25:
When you cloned Apollo repository did you clone with `--recursive` argument to have all the submodules inside?

Can you please check `modules/map/data/san_francisco` and `modules/calibration/data` folders to see if they are empty or not.

If they are empty, first update submodules with `git submodule update --init` then restart Apollo, no need to rebuild it.

alchemz, 2019-01-28:
Hi Martins, thank you for the reply, I tried to update the module by git submodule update --init, but I got the error message 
``
> Unable to checkout '21edeb97e353acaf613b111d25e2c72320395f84' in submodule path 'modules/calibration/data'

martins-mozeiko, 2019-01-28:
Oops, that's our mistake. Forgot to push latest commits in lgsvl/apollo-vehicle repository. I've updated it now, please try again.

alchemz, 2019-01-30:
[image]

alchemz, 2019-01-30:
It is working  now! @martins-mozeiko 🍕 for you on my treat.

closed 2019-02-19 by huyichen19950417