lgsvl/simulator #81: How to setup the initial position?

wsp50317, 2019-01-29:
I tried to setup the candidated vehicle's position and rotation, but it didn't work.
So I tried to find which process clone the vehicle and give it the initial position value, but I couldn't find it out.
Thanks in advance.

martins-mozeiko, 2019-01-30:
If you are running simulator directly in Unity editor by opening SF scene, then starting position is set by `XE_rRgged-apollo` game object. Just move/rotate to place where you want it to start.

If you are running simulator from main menu then start position in SF scene is determined by position of `spawn_transform (1)` game object. Find it in the scene hierarchy and move to place where you want car to spawn. Then save the scene. Now if you will run simulator from main menu, car should be placed in new position.

wsp50317, 2019-02-12:
Thanks! It works.

closed 2019-02-12 by wsp50317