lgsvl/simulator #86: Export Point Could from the stand alone simulator?

bigsheep2012, 2019-02-12:
Hello lgsvl team,

Thanks for sharing this work. 
I have checked the docs, but found nothing about extracting the lidar point cloud.
May I know if it is possible to save the point cloud to a local disk with a stand alone simulator?

Many thanks.

hadiTab, 2019-02-13:
@bigsheep2012 at the moment there is no way to download sensor data in standalone mode. You will need to run ros locally and record rosbags that would contain all of the sensor data.
We are working on an API that would allow that in the future.

bigsheep2012, 2019-02-13:
@hadiTab 
I am trying to run the simulator and visualize the point cloud with ros on Ubuntu 16.04:
1) I download the release standalone simulator and run the executable file (simulator).
2) Then I start ros and run ros bridge and ros rviz:
     roscore
     roslaunch rosbridge_server rosbridge_websocket.launch
     rosrun rviz rviz
3) I switch back to the simulator and select free roaming, localhost, SanFrancisco and XE_Rigged-apollo. I noticed that 'connecting' becomes 'connected'. Click on run. I can control the car normally.
4) I switch to rviz, add pointcloud2 and select topic: /apollo/sensor/velodyne64/compensator/PointCloud2
I see nothing in rviz and get errors in the rosbridge console like:
[ERROR] [1550021669.775496]: [Client 2] publish: Cannot infer topic type for topic /apollo/canbus/chassis as it is not yet advertised

I am quite new to ros. Is there anything I am doing wrong?
Thanks in advance.

bigsheep2012, 2019-02-13:
In addition, the link provided for downloading Unity2018.2.4f1 have some issues. In my case, I got download failed alert while I am trying to install Unity with the download assistant.
[image]

hadiTab, 2019-02-13:
@bigsheep2012 about the download, I just tested the link and it worked for me. I'm not sure why you are experiencing issues with it.

For the error you are receiving:
[ERROR] [1550021669.775496]: [Client 2] publish: Cannot infer topic type for topic /apollo/canbus/chassis as it is not yet advertised
This is because you are running rosbridge locally, and not in the docker container we provided. If you only wish to download lidar data you can ignore this but if you want all of the sensors you will need to use the rosbridge provided in the docker image for Apollo which we host. The reason for this is that Apollo uses a custom message type based on protocol buffers for most of its topics.

In the rviz screen you are posting I believe the reason you are not seeing anything is that you have not selected the correct frame. In rviz, under global options change the fixed frame from "map" to "velodyne" (or maybe "velodyne64") and that should solve your problem.

bigsheep2012, 2019-02-13:
Thanks @hadiTab . It works using 'velodyne'. May I know how to extract the annotations as well? I.e, The bounding boxes (locations, size, rotations, etc. with respect to the controlled car)

hadiTab, 2019-02-13:
@bigsheep2012 the simulator will start publishing ground truth data for 3D bounding boxes if you check the box for it in the simulator UI.
You will need to have the lgsvl_msgs package installed to be able to parse the messages. To install it run:
`sudo apt update`
`sudo apt install ros-$ROS_DISTRO-lgsvl-msgs`

bigsheep2012, 2019-02-13:
Thanks for the reply @hadiTab .
One more question, is there a functionality that could let the player car itself move (like autopilot mode) along a specific path or randomly in the city such that I do not need to drive the car manually for obtaining more lidar data?

hadiTab, 2019-02-14:
@bigsheep2012 unfortunately we do not have such a mode yet.

bigsheep2012, 2019-02-14:
Thanks for your patience. Close the issue.

closed 2019-02-14 by bigsheep2012