lgsvl/simulator #87: Steps to Use Own Apollo Image

moenasrallah, 2019-02-15:
Hello LG SVL Team,

Would it be possible to document the necessary steps to make another Apollo image work with your simulator? Things like rosbridge, nvidia, etc. A guide would help others who have modified Apollo code connect to the sim properly for validation. 

Thank you

frontw, 2019-02-19:
Hi!
Here are steps helped me with r3.0.0 Apollo:

Add ros_pkgs to apollo folder
Add rosbridge script to scripts
```
sudo su
PATH=/usr/local/miniconda2/bin:$PATH
conda install protobuf
conda install pymongo
conda install Pillow
conda install twisted
```

After that you could build it:
```
cd ros_pkgs
catkin_make
```

Copy configs for car and hdmaps
```
modules/calibration/data/LG_SVL_1080
modules/map/data/san_francisco
```

Also you need to modify cameras code to listen to compressed topics.

hadiTab, 2019-02-20:
@moenasrallah all of what we have done can be found by looking at our commit history in our [Apollo fork](https://github.com/lgsvl/apollo). All of our commits will be at the top (after the Apollo commits).
A lot of the commits you will see there will be related to changing configs so they match our vehicle or map better.

The main required component is our [rosbridge](https://github.com/lgsvl/rosbridge_suite). The modifications to it include support for protobuf messages required for Apollo as well as some performance improvements.

The Apollo image we host is built starting from an official Apollo docker image. The dockerfile is actually included in the repository ([here](https://github.com/lgsvl/apollo/blob/simulator/docker/build/dev.x86_64_nvidia.dockerfile)). Lines 14-34 cover the parts related to using nvidia docker. Other modifications to the docker image include applying a patch to PCL to allow it to support older non-avx compliant CPUs and some stuff that allow the visualization tools in Apollo to work (i.e. glfw).

Changes to Apollo are very minimal. We have separate submodules containing our car calibration and configs ([here](https://github.com/lgsvl/apollo-vehicle)), and our hdmaps ([here](https://github.com/lgsvl/apollo-simulator-map)). I believe most of the remaining changes are in the config files of the individual Apollo modules.

One additional change that comes to mind at the moment is for Perception to work in areas where the HDMap is not complete. For this you need disable the HDMapROIfilter in the perception module as seen [here](https://github.com/lgsvl/apollo/commit/d4e2abee7cdd8d2d860a53aaa897f08ed8a8165a), and comment out some code as seen [here](https://github.com/lgsvl/apollo/commit/e252f901b72881860d9c6cce5e493620ee2476c5).

I think that should cover most of it.

moenasrallah, 2019-02-20:
Thank you @frontw and @hadiTab

closed 2019-02-20 by moenasrallah