lgsvl/simulator #88: Are you missing an assembly reference?

amontserratAG, 2019-02-20:
Hi, I am following your Steps for starting simulator in Unity Editor and I got stocked in step 5: "Run the simulator. On the bottom left in the Project panel, Navigate to Assets->Scenes. Then double-click "Menu" (with the Unity icon next to it) to load the scene. At the top of the editor, click the Play button to start the simulator."
Because after clicking the Play button I got the following errors:

Assets/Scripts/ApolloAutoMode.cs(2,7): error CS0246: The type or namespace name `WebSocketSharp' could not be found. Are you missing an assembly reference?

Assets/Scripts/Ros/Bridge.cs(12,7): error CS0246: The type or namespace name `WebSocketSharp' could not be found. Are you missing an assembly reference?

Assets/Scripts/Ros/Bridge.cs(38,9): error CS0246: The type or namespace name `WebSocket' could not be found. Are you missing `System.Net.WebSockets' using directive?

Assets/Scripts/Ros/Bridge.cs(114,37): error CS0246: The type or namespace name `CloseEventArgs' could not be found. Are you missing an assembly reference?

Assets/Scripts/Ros/Bridge.cs(129,37): error CS0246: The type or namespace name `ErrorEventArgs' could not be found. Are you missing `System.IO' using directive?

Assets/Scripts/Ros/Bridge.cs(148,39): error CS0246: The type or namespace name `MessageEventArgs' could not be found. Are you missing `UnityEngine.Networking.PlayerConnection' using directive?

Assets/Plugins/GoogleAnalyticsV4/GoogleAnalyticsMPV3.cs(230,17): warning CS0472: The result of comparing value type `float' with null is always `true'

Assets/Plugins/GoogleAnalyticsV4/GoogleAnalyticsV4.cs(125,19): warning CS0618: `UnityEngine.Application.RegisterLogCallback(UnityEngine.Application.LogCallback)' is obsolete: `Application.RegisterLogCallback is deprecated. Use Application.logMessageReceived instead.'

Assets/Plugins/GoogleAnalyticsV4/AnalyticsManager.cs(33,18): warning CS0414: The private field `AnalyticsManager.sendLaunchEvent' is assigned but its value is never used

Assets/Plugins/GoogleAnalyticsV4/AnalyticsManager.cs(43,17): warning CS0414: The private field `AnalyticsManager.sessionTimeout' is assigned but its value is never used

Any help will be appreciated.

martins-mozeiko, 2019-02-20:
This error probably means that files under [Git LFS](https://git-lfs.github.com/) storage are not downloaded correctly. We use Git LFS for large file storage to improve performance of cloning.

Did you have git lfs installed before cloning repository? If not, please try installing it and repeat the git clone process.

amontserratAG, 2019-02-21:
You were right! thanks a lot.

closed 2019-02-21 by amontserratAG