lgsvl/simulator #91: About lidar

yumianhuli2, 2019-02-22:
[image]
The lidar points is strong far from the host car,how to solve it?Thank U!

martins-mozeiko, 2019-02-22:
Hi.

Could you please tell a bit more what configuration are you running?

* Is this from Editor or binary build? If this is built from source code, then which git commit are you using?
* What configuration of car are you running? Which sensors/checkboxes did you enable?
* How did you move camera further away from car?
* Are you running Windows or Linux? What kind of GPU do you have?

yumianhuli2, 2019-02-22:
HI! 
1 Editor mode. 2019.1 master
2 xe-rigged-apollo,lidar togglebox
3 editor mode.
4 linux,nvidia gtx1080

martins-mozeiko, 2019-02-22:
This is very strange behavior...

Do you have modified any properties of lidar? Ray count, angles, position?
What about default position of camera - when you start map and do not change anything, how does lidar look then?

yumianhuli2, 2019-02-23:
1 I have just changed ray count 32 to 64 .
2 Because this picture was showed in the Editor mode!

martins-mozeiko, 2019-02-23:
Hmm. What happens if you do not change camera position & ray count? Does lidar look correct then?

yumianhuli2, 2019-02-23:
It looks correct if I do not change anything! But if I look lidar in the Scene Mode , the point cloud is strange as my screenshot! Though it looks strange but in fact the point cloud is still right?

martins-mozeiko, 2019-02-27:
It may be that there is some bug related only to visualization in Editor. Not sure though what kind of bug, as I cannot reproduce it. Maybe you could verify that point cloud shows up correctly in rviz?

martins-mozeiko, 2019-02-27:
@yumianhuli2 We found potential issue that may cause this. I think this can happen when you resize (or maximize) game window in Editor on Linux.

We will push latest commits a bit later. For now you can try applying patch I'm attaching here and check if this fixes the issue you are seeing. Patch: [3796341b928a77bef01913dfcf1468ca91104090.diff.txt](https://github.com/lgsvl/simulator/files/2912544/3796341b928a77bef01913dfcf1468ca91104090.diff.txt)

martins-mozeiko, 2019-04-26:
Closing issue as this is fixed.

closed 2019-04-26 by martins-mozeiko