lgsvl/simulator #92: About apollo hd map

yumianhuli2, 2019-02-22:
Hello!
I have real world apollo hd maps and some unity export hd maps.My question is where is the interface between hd maps and apollo algorithm?

martins-mozeiko, 2019-02-22:
Hello.

Are you asking about code that exports map annotations from Unity to Apollo map format? The code for this is available here: https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Map/HDMapTool.cs

yumianhuli2, 2019-02-23:
Thank U!
HI!
I mean how to let apollo algorithm know where is the hdmap? For example,I already have unity export hdmaps and real world apollo hdmaps and I do not know how to use them let apollo algos know them!Thanks!

martins-mozeiko, 2019-02-23:
To export modified map to Apollo, you need to do following:

1) Find `HDMapTool` object in the scene hierarchy and select it.
2) In Inspector press `Export HD Map` - this will create `base_map.txt` file in `hd_map` folder
3) Place base_map.txt file in `apollo/modules/map/data/san_francisco/` folder
4) Inside Apollo docker run `generate_map.sh san_francisco` command - this will convert .txt file into extra files Apollo needs to load map.

After this you should see new map data in dreamview (restart dreamview, if it does reload new data).

yumianhuli2, 2019-02-24:
Hello!
         Generate_map.sh san_francisco` command - this will convert .txt file into extra files Apollo needs to load map,the convert result is routing_config.pb.txt or anyone else? Thanks!

martins-mozeiko, 2019-02-25:
routing_config.pb.txt is configuration file for Routing module in Apollo. It does not contain map data.

Map data is generated in `modules/map/data/MAPNAME` folder. `generate_map.sh` script is using `base_map.txt` file as input and produces other bin/txt files in same folder.

Apollo uses modified OpenDrive format. I don't think it is possible to convert Apollo map to standard OpenDrive format. They use their own customized map format.

yumianhuli2, 2019-03-19:
Hello,if I have apollo hd map.xodr.where can I put  apollo hd map.xodr in apollo docker ?

yumianhuli2, 2019-03-19:
> routing_config.pb.txt is configuration file for Routing module in Apollo. It does not contain map data.
>
> Map data is generated in `modules/map/data/MAPNAME` folder. `generate_map.sh` script is using [...]

Hello,if I have apollo hd map.xodr.where can I put apollo hd map.xodr in apollo docker ?

dongyy1148, 2019-06-21:
@martins-mozeiko When I press Export HD Map, it creates base_map.bin file in hd_map folder, not .txt file. Does it matter?

martins-mozeiko, 2019-06-21:
This is normal. You need to run generate_map.sh script to create other files from this .bin file.

BrunoEduardoCSantos, 2019-07-03:
@martins-mozeiko Maybe I am not clear about HD map generation. So, I have a couple of questions:
1. Is it possible to use Appolo to convert point cloud registered map into HD Map?
2. Is it possible to use Appolo API instead of UI to do the previous task?
Thanks in advance,
Bruno

daohu527, 2019-07-26:
@brunoeducsantos 
> 1. Is it possible to use Appolo to convert point cloud registered map into HD Map?

According to my understanding, cloud registration is only part of the hd-map. The part of the semantic map still needs people to mark it. You can refer to the simulator's map generation tool to make hd-map, but you need to have the whole map model for game in advance. 

> 2. Is it possible to use Appolo API instead of UI to do the previous task?

I am not sure about the features you mentioned, maybe you understand it wrong, it is the tool provided by the simulator, not Apollo.

closed 2020-04-07 by EricBoiseLGSVL