lgsvl/simulator #93: Route Editing greyed out

hyunjaecho94, 2019-02-22:
I am having a similar issue with "https://github.com/lgsvl/simulator/issues/43" but unlike there, I have never been able to access Route Editing tab, and even after apollo.sh clean and rebuild as suggested by martins-mozeiko, the problem persists. 

Rosbridge is connected, so I am thinking Localization is not functioning correctly because martins-mozeiko said that "Route Editing should be enabled when car can localize itself on the map". I have turned on GPS and IMU sensors in simulator and then turned on Localization module, but the car still does not show on the map.

Rosbridge logs are outputting "rosbridge unable to load the manifest for the package pb msgs".

martins-mozeiko, 2019-02-22:
Yes, if Route Editing tab is grayed out that means Apollo is cannot perform localization. If Localization checkbox under modules is enabled that means it does not receive Gps and/or Imu messages.

Have you also selected map and vehicle on top right corner? Apollo won't start localize if map is not selected.

The error about pb msgs means that bridge cannot convert incoming message from Simulator to ROS message for publishing to Apollo. This should work out of box if you are running our modified ROS bridge under our modified Apollo docker container.

How are you running Apollo? Are you using our docker or you have made your own? What about rosbridge - how are you running it (inside/outside docker?) and are you using the one from our repository?

hyunjaecho94, 2019-02-23:
I made some progress, but still unable to run the simulator. There was a permission issue with my docker, so after resolving that, clean/rebuild, and creating/getting into the docker container (which uses lgsvl docker image), when I try to start apollo by running ./scripts/bootstrap.sh, I get two permission errors: apollo/log/rescore.out and apollo/data/log/voice_detector.out. I tried to search for the solution but was unsuccessful. 

Screenshot of error output attached:

[image]

martins-mozeiko, 2019-02-23:
Can you check permissions on `~/apollo/data` folder on your host?
What's the output of `ls -la ~/apollo/data`?

Try stopping docker, completely removing `~/apollo/data` folder (`rm -rf ~/apollo/data`) and the run dev_start/dev_into again.

hyunjaecho94, 2019-02-24:
[image]

Tried removing ~/apollo/data (with sudo) and clean/rebuild but no luck.. The permissions are the same before and after removing and running dev_start/dev_into. Below is the error message I get.

[image]

martins-mozeiko, 2019-02-25:
I think your build was somehow done with different user (using sudo maybe?) and now your normal build cannot succeed, thus it cannot properly generate python files. I suggest to remove whole apollo folder, clone it again and do build in clean space following instructions in readme. Also remove `~/.cache/bazel` folder on your host machine where Apollo is storing build artifacts.

hyunjaecho94, 2019-02-25:
I have removed the apollo folder and cloned again. Now, I can build with no problem, but "Unable to load the manifest for package pb_msgs. Caused by: pb_msgs" error still persists. Additionally, whenever I turn on IMU in the simulator, I get "Cannot infer topic type for topic /apollo/sensor/gnss/imu as it is not yet advertised" error on rosbridge..

martins-mozeiko, 2019-02-26:
Hmm, this is very unusual that it cannot find pb_msgs. Error message for IMU is directly related to this. If pb_msgs is not found the bridge won't be able to convert message from simulator to Apollo protobuf format.

Can you please check what is `PYTHONPATH` environment variable inside docker container?
For me it is `/usr/local/lib/python2.7/dist-packages:/apollo/py_proto:/usr/local/apollo/snowboy/Python:/apollo/modules/tools:/home/tmp/ros/lib/python2.7/dist-packages`

Also what is inside `'/home/tmp/ros/lib/python2.7/dist-packages/pb_msgs` folder?
For me it shows following:

    $ ls /home/tmp/ros/lib/python2.7/dist-packages/pb_msgs/            
    __init__.py  __init__.pyc  msg

hyunjaecho94, 2019-02-26:
PYTHONPATH command is not found inside my docker container even though I am using the docker image provided by the simulator, if they are related. (I am new to docker.) Also, I am missing one file in the pb_msgs folder, as shown below. Could they be causing my errors?

[image]

martins-mozeiko, 2019-02-26:
This looks correct. *.pyc file is generated automatically when Python uses *.py file first time.

Please run `echo $PYTHONPATH` to see value of PYTHONPATH variable.

hyunjaecho94, 2019-02-26:
Looks like it's the same as yours.

/usr/local/lib/python2.7/dist-packages:/apollo/py_proto:/usr/local/apollo/snowboy/Python:/apollo/modules/tools:/home/tmp/ros/lib/python2.7/dist-packages

martins-mozeiko, 2019-02-26:
That is a ROS topic name. They don't point to files or folders. It is just an arbitrary string.

Everything seems to be correct. I have no idea why there could be error when using pb_msgs in Python.
Can you try manually importing them in Python:
1) run `python`
2) enter `import pb_msgs`
3) verify its location, enter `pb_msgs.__file__`

What is the output of this?

hyunjaecho94, 2019-02-26:
/home/tmp/ros/lib/python2.7/dist-packages/pb_msgs/__init__.py

martins-mozeiko, 2019-02-26:
This means that Python knows about pb_msg and can use it.

How are you running rosbridge? What exactly are you executing?

hyunjaecho94, 2019-02-27:
I follow the instructions on lgsvl/apollo github. 1) Create a docker container (./docker/scripts/dev_start.sh) 2) Get into the container (./docker/scripts/dev_into.sh) 3) Start Apollo ( ./scripts/bootstrap.sh) 4) Run rosbridge outside of the docker container ( ./scripts/rosbridge.sh). Apollo and rosbridge were built several days ago (./apollo.sh build_gpu) and (catkin_make).

martins-mozeiko, 2019-02-27:
You need to run rosbridge command inside the docker. rosbridge needs ROS libraries and ROS code that Apollo provides to be able to publish Apollo messages. Please run `./scripts/rosbridge.sh` inside the docker.

hyunjaecho94, 2019-02-27:
I am unable to run the command inside the docker container. I am away from work now, but I think I encountered rosbridge command not found error when I tried to run the command inside. Also, does that mean that catkin_make should also be run inside the container?

martins-mozeiko, 2019-02-28:
Yes, that is right. You need to run catkin_make inside docker. It will build rosbridge package and install it into ROS environment.

We don't really support scenario where bridge runs outside docker, but technically it should be possible as long as you build and install Apollo ROS packages outside. Not sure how well Apollo supports this out of the box.

hyunjaecho94, 2019-02-28:
[image]

When I ran catkin_make inside docker, I got 2 CMake errors, shown above. After some web searching, i ran pip install catkin_pkg, which returned me a different error, shown below. 

[image]

[image]

Note that CMake is trying to use CMake of another user. I am afraid that if I try to change the path, it will get confused. How can I run catkin_make within docker without these errors?

martins-mozeiko, 2019-02-28:
You should not install custom catkin packages. it will probably mess up Apollo setup. They have hardcoded a lot of thing inside docker. And if you install something with pip, apt or ros, you'll need to adjust environment correctly for it to work. We really don't support it, it would be more of question to Apollo developers if you want this.

Did you run catkin_make on clean checkout or on your previous files? Because previously you run it outside docker, so it most likely cached some paths or libraries from your host system. You need to clean ros_pkgs folder so it looks the same as clean git clone and then repeat catkin_make in clean place.

hyunjaecho94, 2019-03-01:
After deleting and re-cloning the repository `(git clone --recurse-submodules git@github.com:lgsvl/apollo.git`), I was able to run `catkin_make` inside ros_pkgs successfully. Then I ran rosbridge inside docker,  and now Route Editing is enabled. However, the perception toggle keeps switching off itself, although I can drive from point A to point B now. Is this a normal behavior?

martins-mozeiko, 2019-03-01:
If it toggles off that means perception process either exits with error or just crashes.

What GPU are you using? Apollo supports only GTX 1080. It probably will run also fine on other 10* GPUs, like 1070, 1060, ... but they do not support anything newer or older.

Check the percepeption log file in `apollo/data/log` folder. Look into perception.out and perception.ERROR files. Maybe these files will provide information about why perception is exiting or crashing.

hyunjaecho94, 2019-03-02:
Okay, thank you so much.

closed 2019-03-02 by hyunjaecho94