lgsvl/simulator #94: Highway Scenario - Apollo Integration Issues

patrickpa, 2019-02-26:
Hello,

I'm working in a project from IVEX to implement a Highway scene into LG simulator (currently it is on my GitHub fork of your repo. https://github.com/patrickpa/simulator) and we are having some issues to integrate it with Apollo, with the NPCs driving in high speed and to properly set the environment. 

Currently, the project is already building and we are using it with Apollo, but we have some issues to properly set up it:

   - The Highway scene is in Scenes folder (highway.asset), currently, just the left lane is working to test the Apollos agent and the NPCs. We are using the EasyRoadsv3 asset to generate the roads, but I think it will not be an issue, we can solve the road format. All the map is annotated, but the lanes are not connected yet;

   - For some reasons, Apollo can’t control the car properly yet. The scenario looks okay and all the information is correct. The trajectory for the ego car is planned correctly for the ego car, but the car can’t follow the trajectory. One hypothesis is that rosbridge is too slow to control the car on simulator when it is running at high speed (100-120kmh);

   - The NPCs as well are having some issues to control the car at the road at this speed, they start to make a zigzag path. Also, we are trying to control the spawn ratio of them, and now we just could control the speed and the amount of spawned NPCs.

Thank you in advance. Best! :smiley:

martins-mozeiko, 2019-02-26:
Hi.

Are you using regular driving mode for Apollo? It is meant only for low-speed city driving. It won't work if car is driving very fast. For highway you should enable navigation mode. It won't use HD vector map at all. It will navigate only using lidar & camera. Following lane will be done by recognizing lane markings from camera.

NPCs probably have similar issues with high speed. We are currently working to completely replace NPC system so they drive more reasonably.

patrickpa, 2019-02-26:
Thank you for your reply. :)

Oh, I see. We'll test the navigation mode. For the NPC system, what could be a reasonable speed for them closer to highway speed?

martins-mozeiko, 2019-02-27:
Probably around 50km/h max to use. One thing you could try is to increase waypoint count - so they are more closer to each other. But not sure if it will increase stability.

patrickpa, 2019-03-07:
So, thanks for the advice, increasing the waypoint counts sort of increased the stability a little, but we still have some issues with the NPCs control.

Another question, we are currently working on a way to pre-define the path of the NPC. Some way to make them follow a path described by a file or something like this. As you said that you are refactoring the NPCs, are you working in something like this?

EricBoiseLGSVL, 2019-03-11:
Hello,

We are working on multiple ways to pass the way point system to NPC's.  You can see this behavior in NPCControllerComponent.cs SetLaneData(List<Vector3>).  This is still heavily based on the MapLaneSegmentBuilder class but we are in the process of making this easier.

patrickpa, 2019-03-13:
Hello, thanks for your answer.

Yes, I saw that and I used it to implement a way to pass a path using a JSON file. It is working quite well right now, but it still needs some improvements.

So, I saw the improvements that you guys made in the wheel movement and it allowed us to use a higher speed, like 60km/h - 80km/h. Thanks! :D

closed 2020-04-07 by EricBoiseLGSVL