lgsvl/simulator #95: Can not build rosbridge for Apollo

Felixgithub2017, 2019-02-27:
I have strictly followed the instructions in the LG Silicon Valley Lab Apollo Fork, however I am not able to build the rosbrige, it always tell me: bash: catkin_make: command not found
[image]

Then I tried this command: source /opt/ros/indigo/setup.bash
I got the following error:
[image]

I don't know what happens, I just followed your instructions carefully.

martins-mozeiko, 2019-02-27:
I do not think Apollo supports running dev_into.sh as root. It probably does not set up ROS environment correctly for root user, it is set up only for non-root user.

Please try running `./docker/scripts/dev_into.sh` script without sudo. Note that you might need to change owner for some files, as you already did build with root so non-root user will not have permissions to write/modify some files.

Felixgithub2017, 2019-03-01:
> I do not think Apollo supports running dev_into.sh as root. It probably does not set up ROS environment correctly for root user, it is set up only for non-root user.
>
> Please try running [...]

I tried one more time and I got the following results:
[image]

[image]

Do you have any suggestions? Thank you in advance!

martins-mozeiko, 2019-03-01:
This means that some files before were created as root user (because of sudo). And now normal user cannot write to these files or folders.

You need to find out which files or folders are owned by root and `chown` them to your own user.

Check ~/.cache folder and ~/.cache/bazel* folders. And also check any folder in ~/apollo. Try running `find ~/apollo -user root` which will list files owned by root.

closed 2019-06-13 by martins-mozeiko