lgsvl/simulator #96: Rvis automatically disappear after running for a few seconds

Felixgithub2017, 2019-02-27:
I am following the instructions of Autoware with LG SVL Simulator, everything seems fine, however when I click the Localization tap, I think there should be a time displayed to the right of  "Ref", but it just doesn't show up.

One more problem, when I click the Rviz button, it just open and running for a few seconds and then disappear automatically.

This always happens, I tried many times.

[image]

hadiTab, 2019-02-27:
The time displayed to the right of Ref doesn't show up until localization has a good estimate. You may need to drive for a while or you may be having other issues with localization. It would be a bit easier to debug this in rviz.

Do you get any error messages when rviz crashes?
What if you try launching rviz manually from a terminal using `rosrun rviz rviz`?

Felixgithub2017, 2019-03-01:
> The time displayed to the right of Ref doesn't show up until localization has a good estimate. You may need to drive for a while or you may be having other issues with localization. It would be a [...]

This is what the terminal looks like after "RViz" is clicked, then it crashed
[image]

[image]

Then I opened another terminal, tried "rosrun rviz rviz", I got the following:
[image]

The followings are the output of terminal after Autoware is started.
This is what the terminal looks like after Autoware is started:
[image]

This is what the terminal looks like after "Map" tab is clicked:
[image]

[image]

[image]

This is what the terminal looks like after "Sensing" is clicked.
[image]

[image]

This is what the terminal looks like after "Localization" is clicked.
[image]

[image]

[image]

This is what the terminal looks like after "Detection" is clicked.
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

This is what the terminal looks like after "Mission Planning" is clicked.
[image]

[image]

[image]

This is what the terminal looks like after "Motion Planning" is clicked.
[image]

[image]

[image]

[image]

This is what the terminal looks like after "RViz" is clicked, then it crashed
[image]

[image]

hadiTab, 2019-03-01:
@Felixgithub2017 It seems like the issue is with the configuration file with rviz (since your rviz is opening normally).
Can you check and see if the config file exists in the directory as shown in your terminal output? This is the directory I'm talking about: `~/autoware/ros/src/.config/rviz/default.rviz`

A workaround for your problem could be to manually launch rviz using `rosrun rviz rviz` and then click on File > Open Config and navigate to the config file.

hadiTab, 2019-03-01:
@Felixgithub2017 I was able to recreate your issue. It seems like there is a bug in the vector map that is causing this. We will be pushing a fix for it to the autoware-data repository soon. In the mean time, you can checkout an older commit.
For example:
```
cd ~/shared_dir/autoware-data/
git checkout 686ed57
```
Once we push the fix you can checkout master again.

tjh49, 2019-03-28:
Hello, @hadiTab 
Do you fix the issue about a bug in the vector map?

hadiTab, 2019-03-28:
@tjh49 the fix has been pushed to master. Are you still having this issue?

tjh49, 2019-03-29:
I have the similar issue .As follows:
1)when I click the Localization tap, I think there should be a time displayed to the right of "Ref", but it just doesn't show up. Can be time message  displayed?  Does this indicate that the lidar positioning(ndt_matching) was not successful in autoware?
2)When using rviz to show vector map,rviz become quite slow,so I don't know how to deal with it .

hadiTab, 2019-03-29:
@tjh49 
1) I believe the number only shows up when the ndt matching algorithm gets a good match. This may or may not happen immediately and really depends on the algorithm and other factors. You shouldn't have localization issues as long as GPS is enabled though since the GPS in our simulator does not have any errors. The point cloud we load as PCD files was artificially generated from the Unity scene and not collected by the simulated lidar. As a result ndt matching doesn't work as well as if the data was collected by the same lidar. You can try building your own pointcloud map by driving around in the simulator and using that for ndt matching instead. If you don't really care about using ndt matching, and just need good localization I would suggest just using the current setup which uses GPS.

2) Unfortunately the Vector Map really slows down rviz and there is not much that can be done about it. You can try unchecking features from the Vector Map so that they are not displayed in rviz, but it will be difficult to generate routes with out displaying the lane centerlines (which are very slow to display in rviz). You may just have to wait a few minutes for everything to load before using it (and it will still be slow).

3) That error is related to obstacle detection using SSD which seems to be crashing. I'm not sure why it is crashing, but it really isn't used for driving around in Autoware so it shouldn't affect anything. You can ignore it for now.

martins-mozeiko, 2019-08-19:
We have verified that our newest simulator release [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) works with latest Autoware release 1.12 without any changes to Autoware code.

Please check that this issue is still happening with Autoware code.

closed 2020-04-07 by EricBoiseLGSVL