lgsvl/simulator #97: perception toggle automatically switches off

hyunjaecho94, 2019-03-02:
Might be related to https://github.com/lgsvl/simulator/issues/21 but I am using the docker image provided by LG SVL. 
On Dreamview, the perception module under Module Controller automatically switches off when I turn it on. Checking perception.out inside "~/apollo/bazel-bin/modules/perception", I noticed that I get "/apollo/bazel-bin/modules/perception/perception: error while loading shared libraries: libcuda.so.1: cannot open shared object file: No such file or directory" error. Running `sudo find /usr/ -name 'libcuda.so.1'`, I get
/usr/lib/i386-linux-gnu/libcuda.so.1
/usr/lib/x86_64-linux-gnu/libcuda.so.1

The reference above suggests that if I am using the docker image provided by LG SVL, I should not have this issue. I am running on Ubuntu 16.04 with Nvidia GTX 1080.

deepaktalwardt, 2019-03-03:
I have the same exact issue. Using the same system configuration and have the same output in perception.out file in apollo/data/log directory. 

Also, I couldn't find nvidia or cuda in the LG SVL Apollo docker image.

martins-mozeiko, 2019-03-04:
Do you have [nvidia-docker](https://github.com/NVIDIA/nvidia-docker) installed as instructed [here](https://github.com/lgsvl/apollo#installing-nvidia-docker)?

Make sure `nvidia-docker` command exists on your docker host. This will allow docker to run image with all necessary drivers (OpenGL, CUDA) working inside docker container.

hyunjaecho94, 2019-03-04:
I must have overlooked the part of installing nvidia-docker. The issue has been resolved. Thank you.

closed 2019-03-04 by hyunjaecho94