lgsvl/simulator #98: GPS Error while using with apollo

deepaktalwardt, 2019-03-02:
I am unable to get GPS data to Apollo's dreamview. 
gnss.ERROR file contains the following log:

> Log file created at: 2019/03/02 15:46:18
Running on machine: in_dev_docker
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
E0302 15:46:18.543507 18809 file.h:106] Failed to parse file /apollo/modules/drivers/gnss/conf/gnss_conf.pb.txt as text proto.
E0302 15:46:18.543723 18809 file.h:145] Failed to parse file /apollo/modules/drivers/gnss/conf/gnss_conf.pb.txt as binary proto.
E0302 15:46:18.546016 18809 apollo_app.cc:60] gnss Init failed: DRIVER_ERROR_GNSS: Unable to load gnss conf file: /apollo/modules/drivers/gnss/conf/gnss_conf.pb.txt

Simulator has all the sensors turned on:
[image]

This is the Dreamview screen. Your help is highly appreciated. Thank you.
[image]

martins-mozeiko, 2019-03-03:
You do not need to turn on GPS, Camera or Velodyne modules in Apollo. These modules (also Radar and CAN Bus) are meant for interface with real hardware devices. They translate incoming data into ROS messages. Because simulator is directly publishing ROS messages, these modules are not needed at all. Enable the GPS checkbox only in simulator and Apollo will receive correct GPS location. As the car in Dreamview is shown in correct position this means Apollo is correctly receiving GPS data from simulator.

When using Apollo with our simulator you need to turn on only Control, Localization, Perception, Planning, Prediction and Routing modules in dreamview interface.

deepaktalwardt, 2019-03-03:
@martins-mozeiko Thank you for your explanation! That is what I thought as well.

Can you explain then how I can control the car in the simulator using Apollo? Using Route Editing or Default Routing doesn't make the car move in the Simulator even though it does move in Dreamview. 

I also receive the Hardware GPS trigger error in Apollo (as you can see in Dreamview). Is that normal? Thanks again!

deepaktalwardt, 2019-03-03:
@martins-mozeiko I was able to control the car in LG simulator using Apollo, however, perception doesn't seem to be working. It keeps toggling back and Apollo cannot see other cars and/or traffic lights. It is similar to issue raised in #97.

closed 2019-03-03 by deepaktalwardt

Martin20, 2020-02-07:
Have you been able to solve your problem?