lgsvl/simulator #100: Car does not restart after stopping from red light or stop sign.

hyunjaecho94, 2019-03-04:
If, at the time of passing, the traffic light is green, the car behaves correctly and continues forward. However, if the light is red or there is a stop sign, then the car stops and does not continue after the light changes to green. I have to manually move the car a little bit forward for it to resume driving autonomously. If there a setting to change this? How can I fix this issue?  Thank you.

martins-mozeiko, 2019-03-05:
One reason for Apollo to stop and not continue is that it does not see traffic light. So it cannot determine if traffic light signal is red or green. Either camera image is not sent from simulator, or Apollo cannot process it correctly.

Please check that in simulator you have turned on both "Main" and "Telephoto" cameras.

When you do that then new window with image from camera should show up when perception module is enabled in Apollo. It will show how Apollo tries to detect color of traffic light. It should look like in this video https://youtu.be/NgW1P75wiuA?t=359 (from 5:58 to 6:07). Do you see this window?

In same video you can see what Apollo thinks traffic light color is - check it the right side. It should switch from RED to GREEN. Is this happening?

closed 2019-03-06 by hyunjaecho94

funpull, 2019-03-26:
I have the same problem,and I do as martins-mozeiko say, still unsolved,I turn on the Main and Telephoto cameras and turn on the perception module in dreamview, but didn't see that window, am I loss something ?