lgsvl/simulator #99: ROS2 bridge

Chizhang1, 2019-03-04:
Dear developers,

It is mentioned in the main page that this simulator also works for ROS2. I am interested in developing something in ROS2 and run it with the LG simulator. Do you have a ros2bridge like the rosbridge in your Apollo fork? Could you give me more information about how your simulator works with ROS2?

martins-mozeiko, 2019-03-04:
Hi.

Yes, our simulator works with ROS2. ROS2 bridge uses exactly same protocol as in ROS1. All you need to do is run different bridge server implementation and simulator will be able talk to it.

For ROS2 we use [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge) implementation. We have run it successfully with our simulator. If you run their rosbridge.js application in same environment as your ROS2 nodes, then you'll be able to send and receive to/from our simulator.

Chizhang1, 2019-03-05:
Hi Martins,

Thanks for you reply! I connected ros2-web-bridge with the simulator. The following topics showed up.

/apollo/sensor/camera/traffic/image_long/compressed
/apollo/sensor/camera/traffic/image_short/compressed
/apollo/sensor/velodyne64/compensator/PointCloud2
/parameter_events
/rosout
/simulator/reset
/simulator/sensor/camera/left/compressed
/simulator/sensor/camera/right/compressed
/simulator/sensor/segmentation_camera/compressed

But when I tried to echo them nothing shows up. I already enabled all the cameras, lidar, GPS and IMU in simulator. I try the same thing with apollo and echo the topic /simulator/sensor/segmentation_camera/compressed. The data showed in the terminal. 

Is there any modification I need to do on ros-web-bridge to make those data be published to those topics?

martins-mozeiko, 2019-03-05:
Are you trying it with Apollo vehicle? That won't work because sensors of Apollo vehicle will publish ROS1 types which bridge probably won't understand and won't be able to convert. You'll need to create/modify code to use ROS2 types.

Alternative is to use "XE_Rigged-lgsvl" vehicle. It is preconfigured to publish ROS2 types.

Chizhang1, 2019-03-05:
Yes, you are right. I used Apollo vehicle yesterday.

I tried "XE-Rigged-lgsvl"  this time. But it looks like there is still nothing published to the topics. I tried ``` ros2 topic echo /simulator/canbus/chassis ```, but I got ``` Could not determine the type for the passed topic```. I tried to echo other topics but nothing showed up.

I tried to use ``` ros2 topic info``` each topic and found there is one publisher for ```/simulator/canbus/chassis``` and ```/simulator/sensor/segmentation_camera/compressed```, one subscriber for ```/simulator/steering_cmd``` and no publisher or subscriber for the other topics.

How did you judge you run ros-web-bridge successfully with the simulator? Are the messages published to the related ros2 topics? Right now I don't run any ros2 node.

Right now the topics are:
```
/apollo/sensor/camera/traffic/image_long/compressed
/apollo/sensor/velodyne64/compensator/PointCloud2
/parameter_events
/rosout
/simulator/canbus/chassis
/simulator/reset
/simulator/sensor/camera/center/compressed
/simulator/sensor/camera/left/compressed
/simulator/sensor/camera/right/compressed
/simulator/sensor/segmentation_camera/compressed
/simulator/steering_cmd
```

martins-mozeiko, 2019-03-05:
If you run `ros2 node list` then you should see `ros2_web_bridge` node - this is the ros2 web bridge working. And if you see topic list that means simulator is already communicating to bridge.

You don't need to run any other nodes to echo data from topics - echoing topic means that "ros2 topic" executable subscribes to it.

Not sure why there is no data coming through. Data should be published as long as arpropriate checbox in simulator UI is enabled (segmentation camera for segmentation_camera/compressed topic).

Just doing `ros2 topic echo /simulator/sensor/segmentation_camera/compressed` should work if you have sensor_msgs/CompressedImage available in your ROS2 environment. I think you should have it as it is standard package. Please check if you have this message type available. You can test this by publishing it to dummy topic:

    ros2 topic pub /dummy sensor_msgs/CompressedImage '{}'

Chizhang1, 2019-03-08:
Thanks for your great job! I tried on the new release version. Right now I can echo data from simulator topics. Here are all the topics I got:
```
/apollo/sensor/camera/traffic/image_long/compressed
/lanefollowing/steering_cmd
/parameter_events
/rosout
/simulator/canbus/chassis
/simulator/control/command
/simulator/reset
/simulator/sensor/camera/center/compressed
/simulator/sensor/camera/left/compressed
/simulator/sensor/camera/right/compressed
/simulator/sensor/segmentation_camera/compressed
/simulator/sensors/lidar
```
But I have a new question. When I drive the car in the map by keyboard and echo the ```/simulator/canbus/chassis``` topic, only twist.linear.x changes, other data are still 0.
[image]
Do you have any clue why it shows that?

martins-mozeiko, 2019-03-08:
For LGSVL car this message does not give velocity values. It simply gives steering value as x coordinate. Code that does this is available here: https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/CanBus.cs#L173

We use this in our end-to-end lane following training use case: https://github.com/lgsvl/lanefollowing

If you need real velocity values feel free to edit code to assign needed x/y/z values in this message or create a new one.

Chizhang1, 2019-03-11:
Thanks you very much! I think this issue could be closed.

closed 2019-03-11 by hadiTab

adamdbrw, 2019-08-01:
I am working on Unity with ROS2 currently (not through a bridge, but using a CS rcl implementation), basing on ros2_dotnet and unity_ros2 projects. I just got working publisher and subscriber examples with Dashing (there is lots to do yet still). Is this something of interest to you, or is your ROS2 bridge performing well enough?

martins-mozeiko, 2019-08-06:
@adamdbrw It would be interesting to see what kind of performance improvements your approach would give over current ROS2 web-bridge.

Ideally we would like to have similar implementation as CyberRT bridge - small binary that runs in ROS receiving environment that can receive and forward binary messages to actual nodes. Then simulator needs to worry only about encoding messages to binary, but not about understanding ROS nodes/protocol. This way simulator user would not need to install whole ROS environment on their simulator machine, which comes with bunch of other issues (for example, which exact ROS2 release version to use).