lgsvl/simulator #101: How to create a real map efficiently for the LG SVL Simulator?

MouriNaruto, 2019-03-06:
First, I have read the document and know how to create a small map in the Unity Editor.
But I am curious about how to create a map like "San Francisco" scene. It is a large map. I think it may take plenty of time to create if follow the document's way. 
I have a lidar and wonder to know how to create a real map efficiently.

I am looking forward to hearing from you soon.

Thanks.
Mouri

martins-mozeiko, 2019-03-07:
To create map from scratch it involves multiple steps.

First you need to create actual geometry of roads, buildings and obstacles. Typically you use Maya, 3ds Max, Blender or similar 3d software. If you want geometry look good then you apply textures to objects. After that you export everything to fbx format files. This is pretty much how SanFrancisco map was made.

Alternative to creating 3d objects from scratch is to get premade assets from [Unity Asset Store](https://assetstore.unity.com/). There are free options or paid options. Quality varies from asset to asset. So it is really up to you to decide what works the best for you.

When you have all assets ready, you can create empty scene in Unity Editor and import assets. Either load fbx file you created, or load assets from Unity Asset Store. Once you position everything in scene, you need to annotate it with our Map Tool, which is described on our documentation page. After this the NPC vehicles should be able to drive on new map, and Apollo/Autoware should be able to navigate it.

MouriNaruto, 2019-03-07:
@martins-mozeiko Do you have a way to use Lidar's ROSBAG or PCD file to import information to the map editor to help the map editing.

martins-mozeiko, 2019-03-08:
At this moment we do not support such functionality. But we have on our roadmap to allow import external vector/hd-maps. So you would not need to generate annotations manually.

Point cloud importing is something we are also thinking about, but we don't have any specific plans yet.

MouriNaruto, 2019-03-08:
@martins-mozeiko Thank you very much.

closed 2019-03-08 by MouriNaruto

daohu527, 2019-06-11:
@martins-mozeiko  By the way when do you support the feature "import external vector/hd-maps", thx!

martins-mozeiko, 2019-06-11:
@daohu527  Sorry, I have not specific deadline. This will come probably closer towards end of year. At the moment we have different priorities.

daohu527, 2019-06-12:
@martins-mozeiko Can I contribute some code? We need this feature in a few months.

martins-mozeiko, 2019-06-12:
@daohu527 Yes, this would be useful to contribute. We have not started work on this feature yet.
I assume you are talking about taking existing map annotations and importing them into our simulator. But this won't include any 3d geometry structure, right? Just road lane information.
What kind of map format are you planning to import?

Note that we'll have a big change in our project structure to support new Unity's HDRP rendering pipeline. But map annotation format won't change much, we are keeping it more or less the same.

If you want to discuss more on this, please open new issue where we can answer any questions you might have.

daohu527, 2019-06-13:
@martins-mozeiko I open a new issue then. We can discuss there.#199