lgsvl/simulator #102: Do you consider communication with Apollo 3.5?

frontw, 2019-03-07:
As you know, Apollo uses own Cyber RT system instead ROS. 
Do you consider communication with Apollo 3.5?

martins-mozeiko, 2019-03-07:
Yes, we are working on Apollo 3.5 support.
Currently we already have internal build with ability to communicate with Cyber RT. At this moment we are integrating this alongside ROS bridge and will be implementing all necessary changes in sensors code so Apollo 3.5 is able to drive virtual EGO vehicle in our simulator.

krishnatoshniwal, 2019-03-12:
That's amazing! Would it be possible to share the timeline for when the bridge would be available. I'm working on the bridge between Apollo 3.5 and the simulator as well and would love to collaborate or contribute!

martins-mozeiko, 2019-03-12:
I expect that we will push code this or next week that will include all changes in simulator to be able to talk to Cyber bridge. This will include code to send/receive messages and new Apollo 3.5 message types (all of them). Simulator will be publishing binary serialized protobuf messages directly over TCP socket. And bridge will only forward message to appropriate cyber node. Similar for receiving - bridge will get binary protobuf message and without any parsing will send it to simulator over TCP socket. There will be no changes in Cyber code required - just running an extra bridge process.

This push will probably not include necessary changes in simulator sensor code to actually publish new messages, which would be required for Apollo 3.5 to be able to control simulator vehicle. This will come later in next month or so so.

It is interesting that you are considering to work on this. Can you share what approach you were thinking/implementing? I would suggest waiting for our push with cyber bridge implementation as it is almost done. And continuing from there if you want to use simulator with Apollo Cyber.

krishnatoshniwal, 2019-03-13:
Thanks for sharing the timeline, quite excited to try out the bridge! By TCP based bridge, do you mean modifying existing rosbridge to work with Cyber channels instead of ROS topics. I am currently familiarizing myself with the Cyber framework and was thinking to modify the previous bridge to work with Cyber channels. I can definitely continue working on it once you push the code!

> This push will probably not include necessary changes in simulator sensor code to actually publish new messages, which would be required for Apollo 3.5 to be able to control simulator vehicle.

Transferring messages (images and point cloud) from the simulator to Cyber will not yet be supported, is that because they need to be converted to protobuf binary before being passed to the bridge?

martins-mozeiko, 2019-03-14:
We created new bridge, because existing bridge is too ROS oriented and has extra overhead due to websocket/json protocol. New bridge is just raw TCP socket that sends protobuf binary blobs forward and back. No protobuf message unserialization happens on the bridge itself. The bridge will be integrated into Apollo buildsystem - it builds new cyber_bridge executable.

By ""not including" these change I mean that we need a time to implement them in existing code. It will take a few more days, and I'm not sure if this will be ready this week. Conversion will be done by Google protobuf library, we don't need to write any code there. We just need to put data into new Apollo 3.5 proto message data types, call protobuf serialization code and write the result to tcp socket.

krishnatoshniwal, 2019-03-14:
So the bridge would directly publish protobuf data corresponding to the channel it is going in and from there on it would be converted to original format right? I am looking into the location where it is serializing and parsing in Cyber code and if it is even happening or not - I believe it has something to do with message node and message Writer and Reader but I am unable to find the base script because of template specialization code, any pointers?  

That makes sense, thanks!

martins-mozeiko, 2019-03-14:
The conversion to C++ structure will happen on receiver said - in original cyber node.

Deserialization in Cyber happens [here](https://github.com/ApolloAuto/apollo/blob/master/cyber/message/message_traits.h#L158). ParseFrom is  protobuf generated code for each message. Serialization code is in same file - SerializeToString function.

Btw, we have pushed simulator code that implements communication with cyber bridge. You can see code in [CyberBridge.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/CyberBridge.cs) file. Serialization code is in [CyberWriter.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/CyberWriter.cs) file. Nobody currently uses these classes in our github master, as we are currently working to add cyber writer and reader functionality to all sensors.

You can get cyber bridge we wrote in my [apollo-cyber-bridge](https://github.com/martins-mozeiko/apollo-cyber-bridge) repo. We will push full Apollo 3.5 integration a bit later when we'll get all sensors working.

But currently using latest simulator and this bridge repo you should be able to talk to Apollo 3.5. All you need is to generate C# source file for proto message serialization. [protoc](https://developers.google.com/protocol-buffers/docs/csharptutorial#compiling-your-protocol-buffers) can do this.

krishnatoshniwal, 2019-03-21:
Thanks for sharing the source code of the bridge and for your helpful replies, can you please help me with the following questions.

1) How exactly does apollo-cyber-bridge repo client read sensor information obtained from the server? And how is CyberBridge reading data from the simulator's sensors (I don't believe that's implemented yet)

2) Does Serialization and Parsing happen twice (Once when a message is published and read on Cyber channel and the second time when it is encoded and sent to server and decoded at other end)? Is that why you were asking to generate C# proto source files (Simulator side proto files)? 

3) The following pattern of code is a bit confusing, is the length variable is being added in 4 different bytes? I know it has to do something with how bytes are arranged (little endian), but I couldn't find more on this online. 

```
data.Add((byte)Op.AddWriter);
data.Add((byte)(channel.Length >> 0));
data.Add((byte)(channel.Length >> 8));
data.Add((byte)(channel.Length >> 16));
data.Add((byte)(channel.Length >> 24));
```

I guess the following code stitches the data obtained. Is that correct? 

```
buffer[offset + 0] | (buffer[offset + 1] << 8) | (buffer[offset + 2] << 16) | (buffer[offset + 3] << 24);
```

4) I was able to start the server on apollo side. But was unable to find a way to connect CyberBridge.cs client to the server. Am I missing something here?

5) Would it be possible to brief what parts are missing from the bridge?

Thanks again!

martins-mozeiko, 2019-03-25:
In our latest code push you can find answers to your questions.

1. Both bridge instances are created in [RosBridgeConnector](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/RosBridgeConnector.cs#L56) constructor. Each vehicle creates BridgeConnector that manages bridge connection (to reconnect if it drops). And each vehicle knows its type (Apollo, Apollo 3.5, Autoware, ...) so it can pass type for creating proper bridge.

2. All the sensors are set up for Cyber communications. For example, Lidar is creating Apollo 3.5 message and publishing in following location in [LidarSensor.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/LidarSensor/LidarSensor.cs#L507) file.

3. Serialization and Parsing happens only once. Simulator serializes data to protobuf binary buffer. Sends it to cyber_bridge over TCP socket. Bridge only reads topic name and whole message as binary blob (no parsing happens). Then it publishes it to Cyber network. Only cyber node that receives it will do the parsing.

4. That codes writes length of topic (channel) to buffer, before putting topic itself in same buffer. This is needed because on receiver side where code reads topic bytes (as string) it needs to know how long is the topic string.

krishnatoshniwal, 2019-03-26:
Thanks Martins! I was able to connect to cyber bridge on Apollo's side and it does receive messages published from the simulator.

The next thing that I tried was to was try to view vehicle movement in Master apollo's Dreamview. As the map, vehicle and calibration files were needed, I copied them from lgsvl-apollo repo. The relevant folders I identified were

/apollo/modules/map/data/san_fransisco 
Replaced /apollo/modules/calibration/data with that of lgsvl's data

However there were many more commits in lgsvl-apollo and changes to other modules calibrations and source code, so I believe there's more to it. Is it possible to visualize the vehicle movement in Apollo by adding these config files?

luqiang21, 2019-03-29:
@krishnatoshniwal For apollo 3.5 to work with the simulator, you need to 
- enable `transform` module by `mainboard -d modules/transform/dag/static_transform.dag` to publish `tf_static`
- enable `localization` module by using rtk localization: `mainboard -d modules/localization/dag/dag_streaming_rtk_localization.dag` to publish `tf`, or you can just change the configuration file to use `rtk` instead of `msf`. Then you can launch `localization` using the button in dreamview.
- decompress images from simulator by `mainboard -d modules/drivers/tools/image_decompress/dag/image_decompress.dag`

And also remember in Apollo 3.5, to enable planning, you need to make sure perception and predication are also enabled.

krishnatoshniwal, 2019-04-02:
Thanks @luqiang21! While starting the Static Transform Component, I got an error saying that the yaml files for 128 and 16 channel lidars were absent. I started looking into other vehicles or in the vehicle params as provided by Apollo but I couldn't find any. Now I understand that these config files are supposed to belong to the vehicle sensors on the simulator vehicle, is there any way I can get access to them?

Here's the error
Extrinsic yaml file is noe exists:
/apollo/modules/drivers/velodyne/params/velodyne128_novatel_extrinsics.yaml
/apollo/modules/drivers/velodyne/params/lidar16_front_up_novatel_extrinsics.yaml
/apollo/modules/drivers/velodyne/params/lidar16_front_novatel_extrinsics.yaml
/apollo/modules/drivers/velodyne/params/lidar16_rear_left_novatel_extrinsics.yaml
/apollo/modules/drivers/velodyne/params/lidar16_rear_right_novatel_extrinsics.yaml
/apollo/modules/perception/data/params/radar_rear_extrinsics.yaml

luqiang21, 2019-04-03:
Hi @krishnatoshniwal, you only need /apollo/modules/drivers/velodyne/params/velodyne128_novatel_extrinsics.yaml, which can be created by just changing corresponding 64 to 128 using the current vehicle calibration files. You can neglect other yaml files errors.

martins-mozeiko, 2019-04-06:
Apollo 3.5 support is fully pushed to github master.
Instructions how to run Apollo 3.5 are available in our Apollo 3.5 fork: https://github.com/lgsvl/apollo-3.5

I'm closing this issue. For any new comments or questions please open new issue.

closed 2019-04-06 by martins-mozeiko