lgsvl/simulator #103: Lisence Limitations

khaledelmadawi, 2019-03-08:
Hi ,
I would like to know the limitations of using this promising platform in the automotive industry, for example if a company used it to build a testing/validation/development platform for a customer, is that legal?
Thanks for your consideration.

zelenkovsky, 2019-03-08:
Regarding current Simulator limitations – our license is not a “traditional” open source. And there is a reason for that. It is crucial for our project to accumulate value and not spill it away. Maybe you know that one of the most heavy-labor part is content(assets) and we do not want people to take Simulator assets apart (cars, maps and etc) and start using them in their own “simulators”. That would fragment community, reduce the value for people who uses “ugly clones” and our own Simulator as well. Another reason: true “open-source” license does not work well with most of other licenses, which prohibits redistribution “as-is” (check Unity Asset Store or SpeedTree License).

**THIS IS NOT A LEGAL ADVICE (please ask your legal team to read the license)**
In short, this is what we want license to say:

> You can build it, run it, test and validate your AD stack with Simulator. You can generate data and data will belong to you.
> You can modify simulator for your own needs (tune or replace the car, change the map).

What we want people NOT to do:

> Take assets or code away and reuse in another environment
> Re-distribute assets or modified version of simulator
> Run simulator in the cloud as a service

If you are willing to do anything from the list above please let us know, we will simply sign a separate agreement allowing things you need. By default it is not part of the github version of the simulator.

# Now let's consider a few cases:
- You want to use Simulator for your Autonomous Driving(AD) stack development and you need to change it.
- Yes, you can do that.

---

- You are providing services to another company who's doing AD stack.
- You can do adjustment, tweaking, small improvements and share it with this company. As far as it is not considered to be a new feature. If company asks you for a new feature - please let us know, we will provide a separate agreement allowing you to do that. There is a chance we might want to incorporate this feature to original simulator.

---

- You are willing to run simulator in your CI without changes (with minor adjustments).
- Yes, you can do that

---

- You are willing to run simulator in managed cloud offering services to other companies
- You will need to get a separate agreement from us, most probably there could be some royalties, but we are very much flexible. If you are running non-profit or educational resource, we are more then happy to talk about free Simulator offerings.

## Current license is not final and might change, but that is what it is right now.

khaledelmadawi, 2019-03-09:
Thank you for your reply, how to contact you to explain our case, and to see further cooperation between us?

bshin-lge, 2019-03-11:
Hello @khaledelmadawi please reach out to us at contact@lgsvlsimulator.com for further discussion. We look forward to hearing from you.

closed 2019-06-13 by martins-mozeiko