lgsvl/simulator #104: Rosbridge Issue

yangydavid, 2019-03-14:
Hi guys, 

I have some issue with the rosbridge connection between LG simulator and Apollo. I am running Apollo on one machine (**Ubuntu 18.04**) and LG simulator on the other (**WIndows**). The two computers are connected via an Ethernet cable. 

From time to time, the rosbridge goes down with a following error: The '/apollo/control' topic is always missing for some reason. 

> /home/tmp/ros/lib/python2.7/dist-packages/rospy/impl/registration.py:358 _connect_topic_thread() [ERROR] [1552528221.352349]: Unable to connect subscriber to publisher [http://in_dev_docker:34893/] for topic [**/apollo/control**]: unable to requestTopic: [Errno 111] Connection refused
 
I'm wondering if any of you have similar issue before, and what would be the possible solution to it. 

Best,
Thanks!

martins-mozeiko, 2019-03-14:
Does the bridge stop completely or it just shows this error and is still run running?

Usually this error shows up just as information that some node is disconnected so bridge won't be able to send messages. But once node comes back live, then sending messages will continue.

yangydavid, 2019-03-14:
Hello @martins-mozeiko 

The rosbridge is still running while publishing error messages. Especially when we enable Lidar/camera sensors, the whole simulator will be delayed by ~2 minute. We are running Apollo with a 1080 GPU card and I am wondering what would be the root cause that affect the transmission of Lidar data? 

Thanks!

martins-mozeiko, 2019-03-14:
You should check CPU usage of rosbridge process. If it is 100% or more, then it means it cannot process data fast enough, so messages will be queued up.

What is CPU spec you are running bridge? And where simulator runs?
Have you modified lidar laser count or measurement frequency?
If you disable lidar, are images then transmitted in real time?
Also check if ethernet is using 1Gbps connection between both machines. 100Mbps is not enough for Apollo data.

closed 2020-04-07 by EricBoiseLGSVL