lgsvl/simulator #105: Cannot load map

Jihwan-Seo, 2019-03-15:
Hello, I’m a student and I’m trying to run Autoware with lgsvl_simulator. When I tried to use Rviz to view the map, but there is a problem that lanes are not loading. How can I solve this problem? Please anyone help me. Thanks.

### Expected Behavior
Map loaded from data with 734 lanes, and these are visualized in Rviz like the following picture
[image]

 
### Actual Behavior
Map loaded from data with 1 lane, and immediately after that Rviz is closed with the following error (picture)
[image]

 
### Specifications
-	Ubuntu version: 16.04 LTS
-	ROS version: Kinetic
-	Autoware version: 1.10.0
-	Graphics card: NVIDIA RTX 2080 Ti
-	CUDA version: 10.0

martins-mozeiko, 2019-03-15:
This error message about mismatched datatype/md5sum is very strange. How are you running rviz & Autoware? Do you launch both of them from inside of autoware docker container?

martins-mozeiko, 2019-03-19:
You need to check you ROS environment. I'm not sure what is wrong with message type mismatch. This seems more of a general Autoware/ROS issue. Maybe you can try asking this question on Autoware support channels. Maybe @hakuturu583 can help here?

We only support our docker image with Autoware. As we are not developing for Autoware every day, we just set up whole Autoware & ROS environment in docker. Everything else related to Autoware setup is kind of out of scope of simulator.

hakuturu583, 2019-03-19:
@martins-mozeiko I think @Jihwan-Seo's Autoware LaneArray message and Autoware Lane Array message in your simulator is not matched.

hakuturu583, 2019-03-19:
@Jihwan-Seo I see.
local_trajectories topic type was changed, but Autoware rviz file still requires marker type.
So, it was conflicted.
If you use rviz file in Autoware master or develop branch, can you post issue to the Autoware??

closed 2019-08-19 by martins-mozeiko