lgsvl/simulator #106: Help me with path planning  in Autoware function

ehduk90, 2019-03-15:
I want to use  path planning function to autoware_car on LGsimulator by autoware.
I followed below steps for it.

input:
   cd autoware/docker/generic
   ./build.sh kinetic

result:
   [image]

input:
		cd ~/autoware/docker/generic
		./run.sh kinetic	

result:		
[image]
[image]
[image]

Click on the 'Map' button for Autoware
[image]
[image]
The error was ignored.

Run Simulator
[image]

Click on the 'Sensing' button for Autoware
[image]

Click on the ‘Enable_LIDAR’ & ‘Enable_GPS’ button for Simulator
[image]
The LADAR has been modified to 64 Chenels and added two cameras.

Click on the ALL button for Autoware
[image]

Specify '2D Nav Goal'
[image]

result(Yellow sign)
[image]

Here has some problem that I couldn't see way point line(Orange) below the capture image unlike other user's case and path planning function doesn't work.
[image]
└source: https://github.com/lgsvl/Autoware/blob/lgsvl_develop/docs/images/readme-rviz.png

computer spec
[image]

martins-mozeiko, 2019-03-15:
I think lane information is not displayed because it is still loading. Our current SanFrancisco map is pretty big for Autoware so it can take many minutes to actually load all map in rviz. Can you try waiting few minutes before specifying 2D nav goal?

Another option is to load our previous vector map map which will have smaller area. You can go to shared_dir folder and checkout one of previous commits from [autoware-data](https://github.com/lgsvl/autoware-data) repository. Try using `93d92d6becc8dbced0a588aadcf0cee18e2c2429` or `e3cfe709e4af32ad2ea8ea4de85579b9916fe516` commits.

ehduk90, 2019-04-03:
No matter where the 2D Nav Goal is, this is the result.
[image]

ehduk90, 2019-04-04:
No matter where the 2D Nav Goal is, this is the result.
[image]
If the 2D Nav Goal is specified in the green area(very short path plannig), the first time the vehicle stops normally after driving.
However,
 1. Try to return to the starting point from the target point.
 2. Look ahead distance is put behind the vehicle.

───────────────────────────────────────────────────────────────
[image]

[image]

[image]

martins-mozeiko, 2019-04-04:
Did you try checking out autoware map from commit I wrote above?
I think your issue is that VectorMap is not loading into rviz. Not sure why... Are you using Autoware from our docker container? Or have you built something else from master (maybe something change there).

ehduk90, 2019-04-05:
Yes, I used the things you suggested.

I've tried  3 versions of "autowere". One version was from LG and the other two, the latest and the older versions, were from autowere site

ehduk90, 2019-04-25:
It's working normally now.
The used version is 824d8f2047caaea 983372cbeb107ea1201509e36c

closed 2019-04-25 by ehduk90