lgsvl/simulator #109: IMU

rukie, 2019-03-18:
When reviewing the IMU code, the IMU ignores gravity in other axes (particularly noticeable when rolling). Should the lines:
```
var linear_acceleration = new Ros.Point3D()
        {
            x = acceleration.z,
            y = - acceleration.x,
            z = acceleration.y + Physics.gravity.y 
        };
```
be modified to:
```
var linear_acceleration = new Ros.Point3D()
        {
            x = acceleration.z + Physics.gravity.z,
            y = - (acceleration.x + Physics.gravity.x),
            z = acceleration.y + Physics.gravity.y 
        };
```

martins-mozeiko, 2019-03-19:
Hello.
Thank you, we'll update this in our code. But I don't think it will give you different values on ROS topic in any situation - currently gravity z & x components are always 0. Normally there should not be situation where you want to set them to non-zero values.

rukie, 2019-03-19:
Ah, I don't think that'll correct what I thought it would. What I meant is that as the vehicle is on a incline, or experiencing heavy roll, the gravity component should get picked up on the other axes of the IMU. 

psuedo code example, not correct: roll pitch yaw = theta phi psi
```
var linear_acceleration = new Ros.Point3D()
        {
            x = acceleration.z - sin(phi)*Physics.gravity.y, // accel due to pitching
            y = - (acceleration.x + sin(theta)*Physics.gravity.y), // accel due to roll
            z = acceleration.y + cos(phi)*cos(theta)*Physics.gravity.y // loss of gravity component due to pitch & roll
        };
```
[image]

martins-mozeiko, 2019-03-19:
Ah, I see. You are saying that IMU should give acceleration in local coordinate system of a car (IMU sensor). Not global world coordinate system as it is doing now. This is a good catch. Thanks! We will fix this.

I guess we haven't seen any issues because we were not using different orientation of car. Are you using simulator for something that makes this behavior noticeable?

rukie, 2019-03-19:
It can help with vehicle pose estimation/orientation algorithms. It can also be used in cruise control for road angle estimates to improve tracking of desired vehicle speed (better torque estimate), among other things. Personally, I'm particularly interested in off-road terrain.

martins-mozeiko, 2019-04-02:
This should be fixed in latest source code on master.

closed 2019-04-02 by martins-mozeiko