lgsvl/simulator #110: About Apollo/Cyber3.5 publish and subscribe topic

yumianhuli2, 2019-03-19:
Is there a example for publish/subscribe topic in lgsvl for apollo/cyber 3.5?
Thanks!

martins-mozeiko, 2019-03-19:
Subscribing & publishing it is pretty much the same as for ROS code. The bridge class interface in same for Ros and Cyber.

To subscribe you do:

    Bridge.AddReader<MessageType>("/topic", msg =>
    {
         // ... use msg
    });

To publish you need to first create writer (in Bridge.OnConnected event callback):

    Writer = Bridge.AddWriter<MessageType>("/topic");

Then do actual publishing:

    var msg = new MessageType();
    msg.Field = Value;
    Writer.Publish(msg);

`MessageType` should be C# class generated by `protoc` from `.proto` file.

As to how create Cyber bridge instance at runtime - we are currently working on it. If you want quick test, you can change this inside RosBridgeConnector constructor. Currently it is hardcoded to RosBridge, but you can replace it with CyberBridge class.

yumianhuli2, 2019-03-20:
ok,thank you!

martins-mozeiko, 2019-04-06:
Apollo 3.5 support is fully pushed to github master.
Instructions how to run Apollo 3.5 are available in our Apollo 3.5 fork: https://github.com/lgsvl/apollo-3.5

I'm closing this issue. For any new comments or questions please open new issue.

closed 2019-04-06 by martins-mozeiko