lgsvl/simulator #111: San Francisco and HDMap rotation

frontw, 2019-03-19:
Hi!

We are trying to create our own scene with HDmap and noticed strange orientation of exported HDmap. 
So we checked again and found that San Francisco HDmap looks good comparing with Google Map, but San Francisco scene looks wrong oriented in Unity3d itself. For example, Market street is horizontal in Unity3d and has about 45 deg incline in Google map and in Dreamviewer.
[image]

So, maybe there is some rotation parameter somewhere?

martins-mozeiko, 2019-03-19:
You are right. There is a rotation for map in [MapUtils.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Map/MapUtil.cs#L456) file which is used when exporting to Apollo HDmap. Currently it is hardcoded, so you'll need to adjust (probably set to 0) in the code.

frontw, 2019-03-19:
> You are right. There is a rotation for map in [MapUtils.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Map/MapUtil.cs#L456) file which is used when exporting to Apollo HDmap. Currently it is hardcoded, so you'll need to adjust (probably set to 0) in the code.

:-)
Thanks!

closed 2019-03-20 by frontw