lgsvl/simulator #112: There is a problem in building LGSVL Autoware Docker image

JackRen88, 2019-03-20:
Hello,everyone!
when I building LGSVL Autoware Docker image,running as follows:
`cd autoware/docker/generic
/build.sh kinetic`
appearing that 
`Step 24/46 : COPY --chown=autoware:autoware ./docker/generic/data/yolov2/yolo.cfg /home/$USERNAME/darknet/cfg/yolo.cfg
COPY failed: stat /var/lib/docker/tmp/docker-builder225186454/docker/generic/data/yolov2/yolo.cfg: no such file or directory`
How should I solve it?

martins-mozeiko, 2019-03-21:
Hello!

Did you clone autoware repository with submodules? `git clone --recursive ...` ?

If not, then you need to update submodules for your autoware repository as we store large data files (like yolo model) in different [repository](https://github.com/lgsvl/autoware-perception).

JackRen88, 2019-03-21:
Hello, @martins-mozeiko 
Accordinhg to instrution I implement as follows:
`git clone --recurse-submodules git@github.com:lgsvl/autoware.git -b lgsvl_develop`
but I have a problem during cloning that 
`Unable to checkout 'bdf6daa0236d59d7898c33e2c55c128a80118c61' in submodule path 'docker/generic/data'`
How should I solve it?

martins-mozeiko, 2019-03-22:
`git clone --recurse-submodules https://github.com/lgsvl/Autoware.git -b lgsvl_develop` is correct command to use. It should work.

Not sure why you see issue with `bdf6daa...` commit. It exists in [autoware-perception](https://github.com/lgsvl/autoware-perception/commit/bdf6daa0236d59d7898c33e2c55c128a80118c61) repository.

Are you running this command in clean space? Without any existing `Autoware` folder in current directory?

JackRen88, 2019-03-23:
Hello, @martins-mozeiko 
I am running this command in  a folder of the home directory.

martins-mozeiko, 2019-03-25:
It seems that error happens when git tries to download caffemodel files in docker/generic/data folder. It says "cannot write data to tempfile ... i/o timeout".

Are you running out of disk space for your user (/home/rcj_robot folder?) Or maybe it is network issue where it timeouts. You need to fix this before building. git clone should be able to download all the files, otherwise build will fail.

JackRen88, 2019-03-27:
Thanks.I now have a Docker image named autoware-kinetic.
when I launch my_map.launch in the my_launch_sf_map folder,there is a error as follows:
[image]
Is there a problem with the file pole.csv in the vector_map_sim_sf_portion folder?

martins-mozeiko, 2019-03-27:
These messages are expected. You can ignore them, map will be loaded correctly.

JackRen88, 2019-03-27:
OK,but when I only launch the  my_map.launch in the my_launch_sf_map folder,and starting rviz to display pointcloud and vector map,rviz gets very stuck.I noticed that point map's size is very small,so it shouldn't make rviz get very stuck

closed 2020-04-07 by EricBoiseLGSVL