lgsvl/simulator #115: git LFS error while git lfs pull

hakuturu583, 2019-03-25:
I try to merge newest version of LGSVL Simulator to our development version.
Then, I was failed with this message.

[d5c1d94c1c08ca216479a4afe26296c82cf7e9aac5a354eb1cd521db999df1ce] Object does n                                                                                                                                                                                                                                                                                           ot exist on the server: [404] Object does not exist on the server

martins-mozeiko, 2019-03-25:
This OID belongs to `Assets/Scenes/SanFrancisco.unity` file. You can check like this:

    $ git lfs ls-files --all | grep d5c1d94c1c
    d5c1d94c1c - Assets/Scenes/SanFrancisco.unity

When I check previous commits for this file I can find commit which made this OID:

    $ git show f3413613b462ff31f80e85ade4d749f9300ca970:Assets/Scenes/SanFrancisco.unity
    version https://git-lfs.github.com/spec/v1
    oid sha256:d5c1d94c1c08ca216479a4afe26296c82cf7e9aac5a354eb1cd521db999df1ce
    size 27128178

If I check f3413613b462ff31f80e85ade4d749f9300ca970 commit in github webinterface you can see and download Assets/Scenes/SanFrancisco.unity file just fine: https://github.com/lgsvl/simulator/blob/f3413613b462ff31f80e85ade4d749f9300ca970/Assets/Scenes/SanFrancisco.unity

What command are you running to get this error?
Can you retry the command, maybe it failed only once?

hakuturu583, 2019-04-02:
When I try "git lfs fetch --all" it shows

```
$ git lfs fetch --all
fetch: 3582 object(s) found, done
fetch: Fetching all references...
[d5c1d94c1c08ca216479a4afe26296c82cf7e9aac5a354eb1cd521db999df1ce] Object does not exist on the server: [404] Object does not exist on the server
[b460bb45e19582a83fcee6468bfe9cf44bf9b692d393965b689cdc8c01faa72c] Object does not exist on the server: [404] Object does not exist on the server
[aab1cf9d13735a051aea3cca95ae89304c2b22eeeede8588b671cc59c5d970e0] Object does not exist on the server: [404] Object does not exist on the server
[bb027f636869c88ec0634b5ae74e61efc5361d51d6100b37bb95de6b516a473c] Object does not exist on the server: [404] Object does not exist on the server
[b75ccc84060c448c305f161329da56aff35568b43f5bedef0b7978bad5bdb206] Object does not exist on the server: [404] Object does not exist on the server
[30106988d83718c28ec0210430fa8d82169e86f0cd2449cca7396ca0387cdae3] Object does not exist on the server: [404] Object does not exist on the server
[5146e9c615080fbaa4e9d120ff74b5e3ae5c4c23fe1f3ebec8900f16cccfe86a] Object does not exist on the server: [404] Object does not exist on the server
[4a8c9c70ac6bc0ac2d30dc499e2216236fcc1f74c7b14ecbaa8763bf3a7524cf] Object does not exist on the server: [404] Object does not exist on the server
[91a40860588920ef33ea7b830fb398b74e6973619be4da894b2219b8b022440a] Object does not exist on the server: [404] Object does not exist on the server
[e555bf760c2342acb6dc7a153ff68784ea7ed1220470793f4084ca01fac6cead] Object does not exist on the server: [404] Object does not exist on the server
[79dee44abd549301954dbe36a38fea96ff3e3a18e982a38ea80cd88916faef16] Object does not exist on the server: [404] Object does not exist on the server
[02ec0690ae05d3cd3c11a199162525e8906a5618eaa4061614a4a223cd3680bf] Object does not exist on the server: [404] Object does not exist on the server
[116fccca0cc94be577cccd8a8d53d67369037d81181a2811c5c474813a1e5d14] Object does not exist on the server: [404] Object does not exist on the server
[2b4fd0002170d27e89fc56a637e15aadc555d463070d3f297a0e1617677ae550] Object does not exist on the server: [404] Object does not exist on the server
[182b7e47143c1cd8d5dfb098d3e0e69def9ebd9a923d07c53489c6ca12a62839] Object does not exist on the server: [404] Object does not exist on the server
[bbf45117b89f57b55b7725acb99ad33939ed7fb331867a381a7d3bca2f0c44c8] Object does not exist on the server: [404] Object does not exist on the server
[0272667b7f91d97a8f31832bcb8abe63452432739a8eb574f787f806f0bde698] Object does not exist on the server: [404] Object does not exist on the server
[edde4b73e2e773159a16eb3117e903c64f5eb0bc44f1805bbdcee32c29cbbf92] Object does not exist on the server: [404] Object does not exist on the server
[63307484090442e9062636d2829e2d33862dc1b934c28bf0396cfaaee2c11807] Object does not exist on the server: [404] Object does not exist on the server
[ac808962ddfadc95b789d216e78813c853f79ac24b237f4b849d8f9cd2e2402e] Object does not exist on the server: [404] Object does not exist on the server
[4b364f9ae5392cd4a5441ef1b15445600e9e3e586e8c126a048d387f6c79d2c0] Object does not exist on the server: [404] Object does not exist on the server
[f230c769f68ba43857126609208046b122a40aeeedc990ae4e8e0f68f7360ff0] Object does not exist on the server: [404] Object does not exist on the server
[671ef70e22aa59585d5d735e756712218eca2c9f5867a19bfddf50eb1db21098] Object does not exist on the server: [404] Object does not exist on the server
[04b658ea258c5f0bc741367ad9458532f7ee8ed429380bff0544388adad38d46] Object does not exist on the server: [404] Object does not exist on the server
[e938eb5635b27c499a1cc37823dcfb3a312d92e1a53239d0f2903db897836822] Object does not exist on the server: [404] Object does not exist on the server
[6f43f527b9425d565e44509e630dc8017b35828cc914d774892ea0dbe8baec79] Object does not exist on the server: [404] Object does not exist on the server
[1b2e14e182a2272140af059e0bf56cb21fb237eb6553e64d29280bfcda885355] Object does not exist on the server: [404] Object does not exist on the server
[75865cc0789df467a014dd52c2f5746be7575b32b2b00b12265a801d608b301e] Object does not exist on the server: [404] Object does not exist on the server
[ec7b06001ba5b26c929b33bab463aeb643b1b8430a9807b68bacd6be0605a711] Object does not exist on the server: [404] Object does not exist on the server
[2e8cb16fe8f7b5be67d744acb02b2c95d42f118bc65e6a7ece1960edd31cb88b] Object does not exist on the server: [404] Object does not exist on the server
[bd8db4c2be94cc3ddbfa70cdfca9444ad2b10a2c0590cbba310491820c6c2fa0] Object does not exist on the server: [404] Object does not exist on the server
[3418c85600689f50e2bc4a1c808cc38f7c6cf46da91a444b38b71aa32619ec3e] Object does not exist on the server: [404] Object does not exist on the server
[b73d2122ed2d1e6199d4d80bb3874f816bb8e6a45a30ea6350ad91a907b9e6c4] Object does not exist on the server: [404] Object does not exist on the server
[11b480c6160df95d3969ae358c67369714cac9953d9f49bf2ff61c44ba99ab21] Object does not exist on the server: [404] Object does not exist on the server
[c89b0ad675a8bb69dc3af4ce92d53caa55ab1a85a1043b2b66036d5587778f35] Object does not exist on the server: [404] Object does not exist on the server
error: failed to fetch some objects from 'https://github.com/hakuturu583/simulator_private.git/info/lfs'
```

martins-mozeiko, 2019-04-02:
How this "simulator_private" was pushed to github? Maybe it did not have Git LFS installed/active, so it did not push files there?

If all these LFS files are from our repository then you should be able to add our remote as another remote. Then fetch all LFS files from our remote. And then push them to your repository.

Basically do following commands:

    git remote add upstream https://github.com/lgsvl/simulator.git
    git lfs fetch --all upstream
    git lfs push --all origin # or whatever name you are using for your remote

closed 2019-06-13 by martins-mozeiko