lgsvl/simulator #116: Obstacle Visualizer Apollo

Andrea2202, 2019-03-25:
Hi,
I'm trying to visualize using apollo perception visualizer, the obstacle detection using only the camera. 
I noticed, however, that the stream for the front_6mm camera is not used. Is it possible to add it?

Thanks

martins-mozeiko, 2019-03-25:
Hello.
We have modified Apollo a bit to be more efficient with current ROS bridge implementation.

We changed "front" camera to read data from "short" camera topic, because we did not want to publish same image two times - we don't simulate separate front and short cameras for performance reasons.

Also we changed image receiving code to directly read compressed image topic (where we send jpg data). By default Apollo subscribes only to uncompressed image topics, which we don't want to send - the data amount is too large.

You can see these changes here: lgsvl/apollo@5a9a8536a21f3a86c863887fcd9e06850424e374

If you are running some other tool/binary from Apollo, then you'll need to apply similar changes there to be able to get camera images directly from simulator.

Andrea2202, 2019-03-26:
Hi @martins-mozeiko ,
Thanks for your answer.
I'm trying yo use apollo perception visualizer because I don't understand if your simulator setup use yolo object detection with camera..
Currently my goal is to able to perform perception with a very slow vehicle using only camera and Yolo's object detection without Lidar and Radar.
Do you know if I can setup your simulator and your customized version of apollo to work only with camera?

Thanks in advance

martins-mozeiko, 2019-03-26:
Our simulator does not do any detection with DNN models at all. We just send camera image to Apollo that does perception on it for detecting traffic lights. We have not modified Apollo perception so I'm not sure what it takes to change perception inside Apollo to use it for object detection. I you'll get better answer on this if you ask to Apollo developers.

Andrea2202, 2019-03-28:
I've modified the apollo perception configuration and now I'm able to use your main camera with apollo DNN and use Apollo offline visualizer. @martins-mozeiko Thanks for your suggestions!