lgsvl/simulator #117: navigation_perception process stopped

funpull, 2019-03-26:
I run the simulator with apollo. everything is good, but when the can drviers to a traffic light it stoped, even the light turns green, it will never go again, and I found the err in apollo console:navigation_perception process stopped! 

Is something wrong with the perception module?  I just turn on perception mode in dreamview.

martins-mozeiko, 2019-03-26:
That's right. If car does not drive past traffic light that means it does not see traffic light because perception module is not sending any messages (as it crashes or exits).

Apollo officially supports perception only on GTX 1080 Nvidia GPU. Are you using this GPU?

One thing to check is log file of perception module. It is inside `data/log/perception.out or perception.ERROR files. Maybe it has some useful error message.

Another thing to check is if  docker can correctly use Nvidia GPU by running `nvidia-smi` command inside docker environment.

funpull, 2019-03-27:
thanks for answer,I'm using GTX 1060, it means the only option is change my GPU? anything else can I do?

martins-mozeiko, 2019-03-27:
1060 should work fine as it is same generation as 1080, but just a bit slower.

Please check the logfile from perception module for any error messages and output of `nvidia-smi` program inside docker.

closed 2019-04-01 by funpull