lgsvl/simulator #118: New Pedestrian path usign Mapth Tool

Andrea2202, 2019-03-28:
Hi,
I'm currently making a new map for simulation.
Everything works fine except the pedestrian path.
Currently I've added MapSegment_PedestrianWalk using MapPedestrianSegmentBuilder(script) and manually adding the target local positions elements.
Unfortunatly the script stall on pedestrianComponents.cs in the function GetRandomTargetPosition.
The line 181 with the function NavMesh.SamplePosition(randomPoint, out hit, 1.0f, NavMesh.AllAreas) return always false.
What am I missing?

Thanks in advance

EricBoiseLGSVL, 2019-03-28:
Hello @Andrea2202 

Have you created a Unity NavMesh in the environment?  Window -> AI -> Navigation
Pedestrians are nav mesh agents and need the nav mesh to navigate the walk-able areas. Once it's baked correctly, there will be a blue mesh on the static meshes that are set to be able to walk.  Let me know if you have any questions setting it up.

-Eric

Andrea2202, 2019-03-29:
Hi @EricBoiseLGSVL ,
thanks for your suggestion.
Perfectly works!

closed 2019-03-29 by Andrea2202