lgsvl/simulator #119: Adding more sensors to the ego vehicle

Shna94, 2019-03-28:
While testing out the simulator with Apollo, I wanted to check if it is possible to create an ego vehicle with a different sensor configuration than the ones shared which send the data to Apollo for on their own channels. I looked up #14 and was able to see what sensors are present on the Vehicles in /Assets/Prefabs/AgentCandidates but was unable to add a new sensor onto the vehicle. I looked up Import New Asset and Import Package but the file opening browser doesn't show any sensor C# scripts in /Assets. Any cues as to what needs to be done afterwards?

Thanks

martins-mozeiko, 2019-03-28:
To add new sensors to vehicle you should create new child object for car (right click on car object and select "Create Empty"). Then select this new object. Then in inspector click "Add Component" and select which script to add. For example LidarSensor for lidar, or VideoToROS for camera sensor. Check existing car model to see what kind of components you need to add for sensors.

After you configure these sensors and their positions you should save vehicle as prefab. And then add this new prefab into AgentCandidates list. This will make it appear in combobox in main menu.

Shna94, 2019-03-29:
Where exactly can I find the Car object? I tried searching for it in the Hierarchy section but couldn't find a relevant object.
[image]

Can you please tell me where exactly I should search for the object?  
[image]

Thanks

martins-mozeiko, 2019-03-29:
It is called "XE_Rigged-apollo".
You can switch to Scene view and then click in 3D view on the car. Unity then will highlight selected object in hierarchy.

Shna94, 2019-04-02:
So the new object was created under the folder of XE_Rigged-apollo (something like a child). However, I couldn't add the object to the menu as shown in [RosRobots](https://github.com/lgsvl/simulator/issues/14#issuecomment-431902431) object in the menu. What should I be searching for in the menu to add the newly created car?

martins-mozeiko, 2019-04-02:
Yeah RosRobots is from a bit older version of simulator. Now it is a bit different - you need to add vehicle prefab to list in `ROSAgentManager` (search it in project tab).

Shna94, 2019-04-02:
So I created the object here 
[image]
Added the Lidar Sensor here
[image]
Added a new element in RosAgentManager here
[image]

But I couldn't find my new Car object in the Project Window. So I dragged the Car object from Hierarchy window to the project window (as I couldn't drag it into the inspector to fill the new space created for the empty element). So now I have it in my Project window

[image]

I am still unable to drag it to fill it in the Element 10 position in RosAgentManager. What am I missing?

martins-mozeiko, 2019-04-02:
Are you creating new car, or just adding new sensor?

If you are creating new car, you cannot do it inside existing car (XE-rigged-apollo). You need to create new top level object. And then add all necessary sensors inside this new object. You'll need to add also extra components on top level object so it knows how to use ROS and how to drive (AgentSetup component and VehicleController, and similar) - checkout how existing car is made, look at inspector to see which components it has. After than save object as prefab - you do this by dragging object from Hierarchy tab into project tab. Then you'll be able to drag this prefab from Project into ROSAgentManager.

If you are adding new sensor, then you don't need to touch ROSAgentManager. Add only new child object to existing car (for example XE_rigged-apollo), and then save changes by pressing "Apply" button in inspector. That's it - now this car will have all modifications you did in Editor.

Shna94, 2019-04-02:
Oh, thanks for the help! I've edited the XE-apollo object from hierarchy by adding another camera sensor 
[image]

The sensor location looks like this, however I am unable to find the Apply button in the inspector window. 
[image]

martins-mozeiko, 2019-04-02:
I think you somehow broke the prefab. How did you add OneMoreCamera to this object?
Did you click on XE_Rigged-apollo in Project view? You should click and modify only XE_Rigged-apollo in Hierarchy tab.

Does car object have Apply button before you added OneMoreCamera? The Apply button should look like in your screenshot in previous comment (where you selected CustomCar).

Shna94, 2019-04-04:
Thank you, I'll just try to create a fresh project and try this again!

martins-mozeiko, 2019-08-19:
With new simulator [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) release you are now able to modify vehicle sensor configuration in WebUI. See documentation [here](https://www.lgsvlsimulator.com/docs/vehicles-tab/#how-to-change-the-configuration-of-a-vehicle) and [here](https://www.lgsvlsimulator.com/docs/sensor-json-options/). If you have any more questions please open a new issue.

closed 2019-08-19 by martins-mozeiko