lgsvl/simulator #120: 'global shutter' camera images

asimonov-hitachi, 2019-03-29:
thanks again for great work developing the simulator!

I am wondering if there is a way to make published camera images like they come from 'global shutter' camera?
At the moment as the car goes faster the rendered objects are blurred, somewhat like it is a 'rolling shutter' camera. It is not quite that. I think it is how Unity renders the moving objects. But do you know if there is a setting for Unity cameras to capture 'as of single moment'?

Kind Regards

martins-mozeiko, 2019-03-29:
Blurry image probably means that your framerate is very low. Our current motion blur effect is a not very good, as it relies a lot on good framerate. The lower it gets the blurrier rendering looks like. I think it works fine for 30-40 fps. In future we'll try to replace it with better implementation.

To disable this motion blur you can check "Low Quality Rendering" checkbox. Although this will disable other camera post-processing effects. To disable only motion blur, you can click on "ScriptableObjects\PlayerCar Follow Cam Post-Processing Profile" asset in Project view, and then uncheck "Motion Blur" checkbox.

asimonov-hitachi, 2019-04-04:
I am running simulator on Dell XPS 9570, which has Nvidia GTX 1050Ti.
For 1024x768 window and Ultra rendering I get the following stats (car is stationary where it has started):
Empty simulator: 70 fps
+Added LIDAR: 65 fps
+Enabled traffic: 64 fps
+Enable pedestrians: 53 fps
+Ground truth 3D: 50 fps
waited few minutes in this configuation: 44 fps
+toggle main camera: 20 fps
+toggle another main camera (see my issue on stereo setup): 10 fps
-toggle another main camera: 19 fps
+low quality rendering: 32 fps
-low quality rendering: did not record
waited a bit: 15 fps

I have disabled 'motion blur' on all Scriptable Objects. even at 14fps I can drive fast and the images are still crisp.

Thanks for helping!

asimonov-hitachi, 2019-04-04:
testing same code on a Linux PC with 1080 (not Ti) gives me about twice FPS than above

martins-mozeiko, 2019-04-04:
That looks like reasonable numbers. Laptop GPUs are weaker than desktop ones. Also 1050 vs 1080 is a significant change.

closed 2019-08-19 by martins-mozeiko