lgsvl/simulator #121: When I launch the Autoware in Docker, I can't open RVIZ, but the process crashes.

fengsiyu, 2019-04-01:
Hello,everyone!
when I enter the  LGSVL Autoware Docker image,running as follows to launch autoware:

cd autoware/ros/
./run

appearing that an dialog box: 
"Failed to run /home/autoware/autoware/ros/run_proc_manager as user root
Unable to copy the user's Xauthorization file."
[image]

After loading all launch files, I press the "RVIZ" button, the window of RVIZ crashed and shutdown as follow.
[image]

Then I saw the command line shows as follows:
"Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface."
"(core dump) rosrun rviz rviz  Segment Fault"
[image]

So, I can't launch the RVIZ to set the goal destination as the instructions.
How should I solve it?

The picture below is the picture of RVIZ in instructions:
[image]

martins-mozeiko, 2019-04-01:
Hello.

I think your Nvidia driver is not installed (or is installed incorrectly). Please check that you can run `nvidia-smi` command on your host and inside docker. It should show available GPU model and driver version.

Or run `glxinfo | grep version` - this should show something like this:

    OpenGL core profile version string: 4.6.0 NVIDIA 418.56

If any of these two does not happen then you need to install driver. For Ubuntu you can gen Nvidia driver here: https://launchpad.net/~graphics-drivers/+archive/ubuntu/ppa

bl4ckrid3r, 2019-07-18:
I'm having the same issue, so I still couldn't figure out the problem. My NVIDIA drivers are working fine, but I still get that same error. Any help ?

daviduhm, 2019-07-18:
@bl4ckrid3r Which Ubuntu version are you running on your host machine? If you are running Ubuntu 18.04, chances are your `Xauthority` file is not in your home directory on your host machine. In [run.sh](https://github.com/lgsvl/Autoware/blob/lgsvl_develop/docker/generic/run.sh), Autoware assumes that your `Xauthority` file is located in `/home/$USER/.Xauthority`, and it will fail to copy it if it's not there. If this is your case, you can try #231.

martins-mozeiko, 2019-08-19:
We have released new version of simulator [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) that works with latest Autoware release 1.12 without any changes to Autoware code. Please check that this crash is still happening with latest Autoware.

closed 2019-08-19 by martins-mozeiko