lgsvl/simulator #127: APIs of LG SVL Simulator

haisongdong-harrison, 2019-04-09:
Hi, I have a tiny experimental ros autonomous driving-project and I want to interface with the LG SVL Simulator. It seems only Autoware and Apollo are officially supported. But I want to talk to the simulator directly from my program.

From the source code, I found a couple of APIs under `Assets/Scripts/Api/Commands` that seem to be pretty powerful, except the commands need to be called via a JSON object of "command" and "arguments" which is different from the `rosbridge` protocol. I guess it's doable if I make a WebSocket client node myself.

So are users supposed to use these APIs? If so where can I find the documentation of them?

martins-mozeiko, 2019-04-09:
We are currently finishing API support, including testing and documentation. It will officially be released sometime in next 2 weeks or so.

While communication will happen via websocket, the actual API will be Python classes/functions. This API is not intended as replacement for ROS or Cyber RT communications to control EGO vehicle. This API is more for controlling environment around you - to place other vehicles in specific positions, make them follow specific waypoints and so on. Main EGO vehicle still should be controlled by ROS or Cyber RT. API is intended to create testing scenarios or for collecting camera/lidar data for DNN training.

You don't need to use Autoware or Apollo as AD stack. As long as you can subscribe to ROS topics (camera, lidar, IMU, ...) and provide control command back any custom ROS/Cyber code will work fine. Good starting point is to take Autoware vehicle as it uses unmodified ROS. You'll only need to remap topic names to match whatever you are already using.

haisongdong-harrison, 2019-04-10:
Thanks for the clarification!
And yeah that was the idea: to manipulate the scenario with the APIs and control the EGO vehicle using ROS.

Just to confirm though, in the LICENSE of this simulator, there are such limitations:
> 1.4 Limitations.  Your use of the Licensed Materials is restricted for use to: (1) build source code or use binaries with ROS1 and ROS2 systems; (2) application to 3D assets; and (3) application only to files created by LG, skipping default assets and code provided by Unity3D as project scaffolding.

Does it mean the APIs also need to accessed from a ROS/ROS 2 node?

martins-mozeiko, 2019-04-10:
We will be changing this part of license - by removing ROS part there.
It is ok to use our simulator without ROS.

haisongdong-harrison, 2019-04-11:
Ah that's a good news! Thank you!

closed 2019-04-11 by haisongdong-harrison