lgsvl/simulator #128: Simulator doesn't sent gear location of vehicle for apollo 3.5

kkhitsko, 2019-04-09:
LGSVL simulator for apollo 3.5 doesn't sent location of gear selector in booth of drive and reverse mode.

martins-mozeiko, 2019-04-09:
I think when we implemented Apollo (3.0 and 3.5) support we were not sure which information about car Apollo actually uses and which not. As it was driving without selected gear information, this was not added. 

Do you think Apollo uses this information and it is important to have it in Apollo messages?

kkhitsko, 2019-04-10:
> Do you think Apollo uses this information and it is important to have it in Apollo messages?

I think it will be usefull to have this information in Apollo messages. There is few Planning and Control algorithms in Apollo that use gear location.

martins-mozeiko, 2019-04-11:
We'll fix this.

martins-mozeiko, 2019-08-19:
Gear selection should be published to Apollo in latest [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) release.

closed 2019-08-19 by martins-mozeiko