lgsvl/simulator #129: Simulator doesn't sent vehicle steering percent for apollo 3.5

kkhitsko, 2019-04-10:
LGSLV simulator don't sent vehicle steering percent if simulator don't receive driving instruction to turn to the left or right.

This makes difficult to debug apollo 3.5 control algorithms using LSSVL simulator.

martins-mozeiko, 2019-04-11:
Do you mean `steering_percentage` field on `/apollo/canbus/chassis` topic?

I can see it being set to correct values when either you are driving manually or Apollo is controlling the car.

In what situation it is not being set?

kkhitsko, 2019-04-12:
> Do you mean steering_percentage field on /apollo/canbus/chassis topic?

Yes

> In what situation it is not being set?

When I drive vehicle manually and don't press left or right button, then steering_percenage field is not represented in  /apollo/canbus/chassis topic.

hadiTab, 2019-04-12:
@kkhitsko the steering_percentage field disappears when the steering percentage is zero. When you are not pressing the left or right key the steering percentage will quickly go back to zero. We implemented it this way to make it easier to drive with the keyboard.

martins-mozeiko, 2019-08-19:
Steering value should be published to Apollo in latest [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) release.

closed 2019-08-19 by martins-mozeiko