lgsvl/simulator #130: I would like to create scenario where

ehduk90, 2019-04-12:
Self-driving vehicle should be surrounded by number of vehicles with different types and size of my choice. This scenario should be helpful for analyze the different on road situation like, traffic jam, vehicles at intersection, blocking by the large vehicles, sudden interruption by the bicycle, motor bike and other small vehicle.
Currently simulator does not allow to modify the number and size of vehicles in the environment.

martins-mozeiko, 2019-04-12:
This will be possible to do with Python API. It will be released with our April release in next few weeks.

ehduk90, 2019-04-12:
> This will be possible to do with Python API. It will be released with our Apri

Thank you very much for your prompt reply.

closed 2019-04-12 by ehduk90