lgsvl/simulator #131: Seems to be there is problem with the loading of vector map.

ehduk90, 2019-04-12:
Even though i have followed the suggestion as mentioned in the previous comment, but still couldn't able to load vector map. #106
Is the vector map you gave normal?

Below are the things I've tried.
	 **Commits**
-93d92d6becc8dbced0a588aadcf0cee18e2c2429
-e3cfe709e4af32ad2ea8ea4de85579b9916fe516
-d71dfdcb2cee3103cc3fd9e25107e7813ac3100d 
	 **Cuda**
-9.0
-8.0
	 **GPU**
-NVIDIA 1070Ti
-NVIDIA TITAN X(Pascal)
**Autoware**
One version was from LG and the other two, the latest and the older versions, were from autowere site.

Is there anything wrong with compatibility?

martins-mozeiko, 2019-04-12:
Can you try with e3cfe709e4af32ad2ea8ea4de85579b9916fe516 commit for autoware-data repository?
I just tried with this one and it loaded smaller part of SF in rviz.

The problem with later commits is that they work, but they are super slow. Depending on computer (CPU, not GPU) it may take few minutes to load all information of vector map. But the map data files itself are correct and usable.

I cannot speak for code from autware github, it may or may not work - as they are modifying autoware and we are currently not rebasing our fork on top of their master.

ehduk90, 2019-04-25:
It's working normally now.
The used version is 824d8f2047caaea 983372cbeb107ea1201509e36c

closed 2019-04-25 by ehduk90

yangjiongfeng, 2019-07-30:
@ehduk90 
I made the same mistake as you.
How did you solve it?
Where can I download your autoware-data files?