lgsvl/simulator #132: Would like to create scenarios with the pedestrian in cross walk which is currently not included in the simulator.

ehduk90, 2019-04-12:
The current simulator doesn't included pedestrian crossing the road in crosswalk. Although there are pedestrian  walking in the sidewalk, but not on the crosswalk. It will be more helpful, if the simulator includes pedestrian on the crosswalk scenarios.

martins-mozeiko, 2019-04-12:
We have plans to make pedestrians walk on the crosswalk. But I don't have more detailed timeline on this feature.

ehduk90, 2019-04-12:
> We have plans to make pedestrians walk on the crosswalk. But I don't have more detailed timeline on this feature.

Thank you very much for your prompt reply.

closed 2019-04-12 by ehduk90

outmanlol, 2019-06-21:
@martins-mozeiko I want to build a senerio that pedestrian walk cross the road ,but the pedestrian can not move on the road area. I dont know why? Is it a bug?

martins-mozeiko, 2019-06-21:
@outmanlol Please see this answer how to modify areas where pedestrian can walk: 
#170