lgsvl/simulator #133: Use lgsvlsimulator to interact on windows

shmily-skx, 2019-04-12:
How to use the data information of the vehicle and sensor in the lgsvlsimulator and control the vehicle motion through the control algorithm on the windows platform.

martins-mozeiko, 2019-04-12:
You will need to have ROS running to control the vehicle. On Windows officially this is possible only with ROS2, because ROS1 is not supported. You'll need to follow [ros2 installation instructions](https://index.ros.org/doc/ros2/Installation/Windows-Install-Binary/). After than you'll need to install rosbridge for ros2 - get it [here](https://github.com/RobotWebTools/ros2-web-bridge). After running the web bridge you should see ROS2 topics that simulator is publishing.

Installation of ROS on Windows is pretty complex. Alternative is to use docker on Windows to run ROS in Ubuntu environment. That will allow to get ROS1 or ROS2 with pretty standard apt-get install commands, including rosbridge. But installation of ROS inside docker is a bit out of scope of simulator project. You should check guides & documentation of docker & ROS to get instructions how to do this.

shmily-skx, 2019-04-12:
> You will need to have ROS running to control the vehicle. On Windows officially this is possible only with ROS2, because ROS1 is not supported. You'll need to follow [ros2 installation [...]

So, I want to get the keyboard control interface now. There are also pose information, radar information, and camera information of the vehicle. I want to write a control strategy that allows the car to implement the ADAS function and the automatic driving function (L3) in the simulation environment.

Do you think it is easy to implement on Windows, or is it recommended to operate on Linux? If you are on Linux, is it only possible to download emulation and ros, but also need to download apollo and autoware?

martins-mozeiko, 2019-04-12:
You should check out our [lanefollowing](https://github.com/lgsvl/lanefollowing) project - it shows example how to create from scratch very simple "AD stack" based on ROS2. This AD example does only lane following from camera image with DNN model. Example code requires running docker setup on Linux with Nvidia GPU.

It is possible to do all that on Windows, but you'll need to first set up ROS2 & ros2-web-bridge on Windows. There are instructions how to install ROS2 on Windows in their documentation. But on Linux it will be simpler as that is primary platform where ROS is developed.

You do not need Apollo and Autoware if you want to talk to our simulator directly. You can use Apollo/Autoware if you want to modify, improve or work only on one specific component in their AD stack (for example, perception).

shmily-skx, 2019-04-12:
> 
> 
> 你应该看看我们 [lanefollowing ](https://github.com/lgsvl/lanefollowing)从头项目——它的例子展示了如何创建基于ROS2非常简单的“广告栈”。 这个广告的例子也只有从相机图像后巷款模型。 示例代码需要与Nvidia GPU码头工人安装在Linux上运行。
> 
> 可以做所有,在Windows上,但是你需要首先建立ROS2 & ros2-web-bridge窗口。 窗户上有说明如何安装ROS2文档。 但在Linux上它将简单的主要平台,ROS。
> 
> 你不需要阿波罗和Autoware如果你想直接跟我们的模拟器。 您可以使用阿波罗/ Autoware如果你想修改,改善或工作只能在一个特定的组件在他们的广告堆栈(例如,知觉)。

Thank you for your answer, I will try it. And feedback the results to you

shmily-skx, 2019-04-12:
> You should check out our [lanefollowing](https://github.com/lgsvl/lanefollowing) project - it shows example how to create from scratch very simple "AD stack" based on ROS2. This AD example does only [...]

Thank you for your answer, I will try it. And feedback the results to you

shmily-skx, 2019-04-15:
> You should check out our [lanefollowing](https://github.com/lgsvl/lanefollowing) project - it shows example how to create from scratch very simple "AD stack" based on ROS2. This AD example does only [...]

I am now connected using rosbridge. I didn't use apollo or autoware, just use ros. When I use the rostopic list, only a few topics are listed here. I can make the car reset by pub /simulator/reset. So, how do I get more sensor information and vehicle motion information now? Also, if I want to use ros to control the car to move, which topic should I use?

Or do I have to install apollo or auto to achieve control of the vehicle?

martins-mozeiko, 2019-04-15:
I think you are using Apollo vehicle in this screenshot? You should first try using LGSVL vehicle (with ROS2) in main menu. Apollo or Autoware vehicles require custom ROS message types to be available. While you don't need to install full Apollo/Autoware it requires a bit of work to compile custom msg files.

With LGSVL vehicle you should be able to move it by publishing `geometry_msgs/TwistStamped` type on `/lanefollowing/steering_cmd` topic.

If you want to use existing Autoware or Apollo vehicles then you'll need to adjust C# code to use only standard ROS message types that does not depend on Autoware and Apollo message types.

closed 2020-04-07 by EricBoiseLGSVL