lgsvl/simulator #134: Use unity build error in windows

shmily-skx, 2019-04-12:
[image]

I don't know how to download these missing files.

martins-mozeiko, 2019-04-12:
We use [Git LFS](https://git-lfs.github.com/) for binary file storage. You need to have git lfs installed before cloning the project. Make sure you run `git lfs install` for git to install necessary hooks for your user.

shmily-skx, 2019-04-12:
> 
> 
> 我们使用 [Git LFS ](https://git-lfs.github.com/)对二进制文件存储。 之前,您需要安装git lfs克隆项目。 确保你运行 `git lfs安装 `为您的用户git安装必要的钩子。 更详细的步骤按照说明的自述: [在编辑器统一步骤启动模拟器 ](https://github.com/lgsvl/simulator/blob/master/README.md#steps-for-starting-simulator-in-unity-editor)。

Thanks for your reply,
My source file was downloaded from github, and git-lfs is already installed on my computer, but I always get an error when I clone it according to the readme.md tutorial. I think this is a problem with the network.
If you can compile in unity, can you export vehicle motion information and sensor information? I want to be an autonomous driving project as my graduation project.

martins-mozeiko, 2019-04-12:
If you used "Download ZIP" link then this will NOT work. GitHub does not support downloading files with LFS storage.

If you want latest source code you need to run "git clone ..." command. What error do you get when cloning repository?

To run simulator you have two options - one is to run binary already compiled by us. You can download these binaries from our releases tab. Second option is to open cloned project and run it from Unity Editor. Running in editor will allow you to change configuration of map & vehicle, but it will run with a bit slower performance. Both options will allow you to receive vehicle position/motion & sensor information over ROS topics.

closed 2019-06-13 by martins-mozeiko