lgsvl/simulator #136: Collecting data from camera and sensor

kk2491, 2019-04-17:
Hi Guys,

Thanks for this wonderful simulator. 

Could you please let me know is there a way I can collect Image data and sensor (lidar) data from the simulator by running it in standalone mode. (not with Apollo or Autoware)

Thank you,
KK

martins-mozeiko, 2019-04-18:
This is possible with our newly released Api. Check the readme in [Api](https://github.com/lgsvl/simulator/tree/master/Api) folder. Small examples how to saving lidar data and camera image are available under [quickstart](https://github.com/lgsvl/simulator/tree/master/Api/quickstart) folder - check the [06-save-camera-image.py](https://github.com/lgsvl/simulator/blob/master/Api/quickstart/06-save-camera-image.py) and [07-save-lidar-point-cloud.py](https://github.com/lgsvl/simulator/blob/master/Api/quickstart/07-save-lidar-point-cloud.py) files.

We are currently writing documentation for this with more details.

kk2491, 2019-04-19:
@martins-mozeiko Thats great. That would be really useful. Just wanted to know if the prebuilt version is released with Python API ? 

Thank you,
KK

shmily-skx, 2019-04-20:
> This is possible with our newly released Api. Check the readme in [Api](https://github.com/lgsvl/simulator/tree/master/Api) folder. Small examples how to saving lidar data and camera image are [...]

This is great, so can I use this API directly in windows or ubuntu? How is the lgsvl library file imported? Still need to use rosbridge?

martins-mozeiko, 2019-08-19:
Yes, API can be used on any platform - regardless of where simulator is running. "lgsvl" library is our Python API module.

rosbridge is needed only if you are interacting with AD stack. API itself communicates directly to simulator.

Martin20, 2020-02-07:
Is that possible to change the camera scene size? If so, where?

Barburat, 2020-02-13:
I'm trying to open the links to the Api readme folder and the quickstart folder but it only leads to a 404 page not found. Is there another way of getting the info?

Martin20, 2020-02-13:
Did you try this:

https://github.com/lgsvl/PythonAPI

closed 2020-04-07 by EricBoiseLGSVL