lgsvl/simulator #137: IMU data timestamp_sec is same for some consecutive messages

frontw, 2019-04-17:
I run LGSVL Simulator from built executable and do 'rostopic echo /apollo/sensor/gnss/imu | grep timestamp_sec' in Apollo on the same machine.
I get data like this:
```
  timestamp_sec: 1555509908.23
  timestamp_sec: 1555509908.23
  timestamp_sec: 1555509908.24
  timestamp_sec: 1555509908.25
  timestamp_sec: 1555509908.25
  timestamp_sec: 1555509908.27
  timestamp_sec: 1555509908.27
  timestamp_sec: 1555509908.28
  timestamp_sec: 1555509908.32
  timestamp_sec: 1555509908.32
  timestamp_sec: 1555509908.32
  timestamp_sec: 1555509908.32
  timestamp_sec: 1555509908.35
  timestamp_sec: 1555509908.35
  timestamp_sec: 1555509908.35
  timestamp_sec: 1555509908.38
  timestamp_sec: 1555509908.38
  timestamp_sec: 1555509908.38
```
Is it rounding issue, or it is not possible to get 100Hz data with proper 0.01s interval?

martins-mozeiko, 2019-04-17:
This is possible to fix. I think we are not using correct timestamps. We'll look into fixing this.

martins-mozeiko, 2019-04-19:
Duplicate of #138.

closed 2019-04-19 by martins-mozeiko