lgsvl/simulator #138: IMU data comes in packets

frontw, 2019-04-18:
We are working on car dynamics identification and use your simulator for this task as far as for other SDC project simulation tasks. 
For dynamics identification it's very important, how reliable is data, it should have constant rate and latency.
Here #137 I mentioned IMU data timestamp_sec is same for some consecutive messages. 
And then I made another test to see, how data actually comes.
I record bag (set /use_sim_time to false) and use rqt to display time, message actually came:
[image]
And, as you can see, about 3-7 IMU msgs comes together in some sort of bunch. So, period is not 1/100.
I guess, there is some problem of buffering in part of Unity or rosbridge. I run simulator and Apollo on the same machine, so it is not network issue.  
You could compare IMU rate with tf, where publishing period is constant.
[image]
[image]

Thanks for the great project!

martins-mozeiko, 2019-04-19:
We'll look into fixing this.

daviduhm, 2019-05-03:
Hi @frontw,

Just wanted to update you that we have fixed this issue on our internal repository, and the fix will be available soon in a week or so after our weekly testing is done. In the meantime, here is the rqt graph showing IMU messages being published in a constant interval:

[image]

frontw, 2019-05-03:
Hi @daviduhm!
It’s super cool, thank you for this update! 
Looking forward to test it after publishing.
Btw, what was the reason of this issue?

martins-mozeiko, 2019-05-03:
Our IMU implementation was tied to Unity physics update step and Unity does multiple physics steps as fast as it can during one frame. We did not notice this a problem because Apollo was driving just fine with original implementation. Now we changed code to produce IMU messages only when their corresponding values (velocity/acceleration) are supposed to be measured - which should give approximately  evenly spaced 100Hz.

frontw, 2019-05-21:
Finally, returned to this and could confirm, that it looks good now.
Thanks!
[image]

closed 2019-06-13 by martins-mozeiko