lgsvl/simulator #139: How to add my own Vehicle Model.

majichang, 2019-04-18:
Hi, LG simulator guys. I am working on Autoware with LG simulator,and I have a problem to solve it, someone can do me a favior!
     When i excute ./simulator in the terminal, in the "Free Roaming" page, how can I add my own Vehicle model in the simulate environment, and can select it  in the Vehicle texttable. very thanks for your help.

martins-mozeiko, 2019-04-18:
If you have already created 3d model and imported it into simulator Unity project, then you need to add the prefab to list in ROSAgentManager object. Search for it in project and add new vehicle to list of available prefabs. Here's a screenshot of how it looks currently:

[image]

Drag your prefab into new element.

lemketron, 2019-05-08:
Check out this new document recently added to the LGSVL Simulator web site that describes [how to add a new ego vehicle](https://www.lgsvlsimulator.com/docs/add-new-ego-vehicle/).

yangjiongfeng, 2019-08-01:
@martins-mozeiko i also want to add my own Vehicle Model.

this is my robot 
[image]
i use solidworks save ase a .stl file.

1.What is the format of your robot model? .urdf or .stl?
2.and i wan to know How do you load the model into unity?
Looking forward to your reply!

EricBoiseLGSVL, 2019-08-01:
@yangjiongfeng 

1) Unity accepts multiple file formats for 3D geometry; .fbx, .dae (Collada), .3ds, .dxf, and .obj files.  I recommend always using .fbx
2) If your model is not one of these, use Blender (free) or Maya (paid) to import file, make sure geometry is correct (center pivot to bottom of model, make sure z forward is set) and then export as .fbx

closed 2020-04-07 by EricBoiseLGSVL