lgsvl/simulator #140: Pre-built model with API - Latest release

kk2491, 2019-04-23:
Hi Guys,

Could you please let me know when the latest pre-built simulator package would be released with Python API functionality.

Thank you,
KK

martins-mozeiko, 2019-04-23:
Python API for simulator was released last week. You can it with v2019.04 binary build or just use master of git repository.

kk2491, 2019-04-24:
@martins-mozeiko I was not able to find ```v2019.04``` in the latest release / download section. Could you please let me know when it would be released?

martins-mozeiko, 2019-04-24:
It is available in releases page: https://github.com/lgsvl/simulator/releases
Here's a direct link: https://github.com/lgsvl/simulator/releases/tag/2019.04-rc1

kk2491, 2019-04-24:
@martins-mozeiko Thats great. Now I am able to download the latest release. Thanks for the help.

closed 2019-05-10 by kk2491