lgsvl/simulator #141: Image publishing not working correctly

christian-lanius, 2019-04-23:
Thank you for your great work!
I am playing around with the latest github commit and your Apollo fork. Using cyber_monitor, the published image rate is somewhere between 40 and 50 frames. Changing the "Camera Publish Rate" in the simulator has no effect on this. I expected that the rate of published images would change, depending on this value, is this a misconception from me?
Using the cyber_visualizer tool, the high refresh rate of the camera can be confirmed. Another problem seems to be, that the images are not published in chronological order. I have attached a screenshot, showing a graph with the relationship between received timestamp (as the timestamp_sec field in the header of the /apollo/sensor/camera/front_12mm/image/compressed channel) and the current time of my pc. I expected the relationship to be strictly increasing, i.e. the images are received in the order of creation. There are also images sharing the exact same timestamp.
[image]

martins-mozeiko, 2019-04-23:
The camera publish rate text field should affect the rate how cyber receives images. It's not always exact due to how GPU works, but simulator will try to be as close as possible. I see that I can change the rate from 5 to 20 and cyber_monitor reflects this change. One thing to note is that actual capture & publish rate will never be higher than half of simulator rendering frame rate (because there is one frame delay to get the image from GPU).

How do you get 40-50 frames for images? Is it happening by default or only when you change this value in UI?

As for timestamps - is this happening for actual images (data arriving in wrong order) or is it only for timestamps? If you enable perception which shows camera image, or "camera sensor" toggle in dreamview - does image jumps forward and back?

One reason maybe could be slow network - are you running this on separate computers? The network between them should be 1Gbps. 100Mbps router/switch is not enough.

christian-lanius, 2019-04-23:
> As for timestamps - is this happening for actual images (data arriving in wrong order) or is it only for timestamps? If you enable perception which shows camera image, or "camera sensor" toggle in dreamview - does image jumps forward and back?    

I used cyber_visualizer to look at the data received by Apollo. There, the same out of order reception can be observed: The images are very jittery, seemingly jumping forwards/backwards in time, while on average (same as the timestamps) still progressing.

> How do you get 40-50 frames for images? Is it happening by default or only when you change this value in UI?

I left the text field on the default value 15, when I observed this behaviour. It does not seem to always occur, after restarting the simulator, the problem did not occur again.

> One reason maybe could be slow network - are you running this on separate computers? The network between them should be 1Gbps. 100Mbps router/switch is not enough.

Running on the same computer, the simulator on the native host and Apollo inside the docker container.

closed 2020-04-07 by EricBoiseLGSVL